PubInv/krake #1: Reconsider using the ESP32 or using the Arduino R4 Wifi

RobertLRead, 2024-04-28:
Dear Team,

Recently I was working on the GPAD paper for HardwareX. I was attempting to control it with a microcontroller as a demonstration. While working on this, I had occasion to order an Arduion R4 Wifi, which is a relatively new product.

It has the advantage of using the origin UNO footprint (similar to the GPAD). It is also expected to be widely known to hobbyists (though the ESP32 is as well.)  I therefore think there might be some minor advantage  in using the R4 instead of the ESP32.

nk25719, 2024-06-02:
I would also like to experiment with the Arduino R4 WiFi. It is available at the local supplier here in Lebanon. If we decide to go with the same enclosure, we can do that.

There is the RYLP998 module available as well, which we could consider.
youtube video: https://www.youtube.com/watch?v=9azEfCQNhSA

ForrestErickson, 2024-11-21:
I think we are far into the development with the ESP32 and learning the R4WiFi system at this point will not get us to a better Krake and will only slow us.

I urge we close this as will not fix.

Learning the Arduino R4 capabilities should be an independent activity.

CC: @RobertLRead

RobertLRead, 2024-11-21:
Will not Fix.

closed 2024-11-21 by RobertLRead