PubInv/krake #2: Add a standard GPIO header port the Krake...

RobertLRead, 2024-04-28:
Dear Team,

Today while trying to work on the GPAD for the HardwareX paper, I released that I had a major misconception: you can't program it through the serial port with another microcontroller with a USB cable, because most USB devices on micrcontrollers (like the UNO) can only act as a USB peripheral, and not as a USB "Most". I should have known this, but was confused about it.

I believe allowing hobby-level interaction with the Krake is extremely important. The GPAD has an RJ12 jack, but without special equipment, such as splitting an RJ12 data cable, or soldering the RJ12 into a special board (and it does not even fit into the standard .1" perfboard standard. 

I therefore suggest that the Krake have a special port (possibly with a cover) that exposes the standard .1" female header format (which can be connected to with standard jumper wires.) These ports should be used to expose the 4 SPI wires, and also a second serial port (NOT a USB), which I think does work from micro-controller to micro-controller.  This will make it easier for beginners to experiment with it with no special equipment.

ForrestErickson, 2024-04-30:
Not sure what you mean by " you can't program it ".  To what "it" do you refer?

RobertLRead, 2024-04-30:
I meant "you can't send alarm commands to a krake with a USB cable connected to most microcontroller USB ports because they are not USB hosts".

ForrestErickson, 2024-10-02:
I think this requirement can be better defined as follows:

Add to the Krake a connector for 3.3V logic level connection. 
The nature of the connection is for: 
1. A four pin 3.3V SPI 
2. A two pin 3.3V UART (TX and RX)
3. Power, 5V, and 3.3V
4. Ground pins

Consider a right angle header (male pins).

ForrestErickson, 2024-10-02:
If the UART port were to match one of the well known USB to UART converters all the better

https://www.amazon.com/FPOUT-FT232RL-Serial-Adapter-Module/dp/B0CMQTWMDL
This would include flow control pins.

Carful review of the power pins is required. It is probably NOT possible to power the Krake through the converter as the Host to the converter will only supply 100 mA.

ForrestErickson, 2024-10-03:
I think 

> "...you can't program it through the serial port with another microcontroller with a USB cable, because most USB devices on micrcontrollers (like the UNO) can only act as a USB "Device" aka peripheral, and not as a USB "Host". 

Should be rewritten as:

> "... a typical micro controller in a medical device cannot send commands  to the Krake through the USB-serial port, because most USB devices on micrcontrollers (like the UNO) can only act as a USB "Device" aka peripheral, and not as a USB "Host".

ForrestErickson, 2024-11-20:
@nk25719 Please review this and lets get something into the schematic.

nk25719, 2024-11-26:
U1RXD and U1TXD are still available (GPIO9 and GPIO10)

[image]

ESP32 tech ref:
https://www.espressif.com.cn/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf

ForrestErickson, 2024-11-26:
@RobertLRead 
If you follow the recent information we added above you will see that the ESP32 has a UART we have not yet used (UART1).
I do no think there is any conflict with using it, but till we write some firmware and test we will not KNOW.
If you want a serial interface it should be a full compliant RS232 interface (RS232 specifies the electrical interface) and on a DB9 Female connector wired as "DCE".

This will add only a little cost for components but may make more extensive design changes to the enclosure. We will not know till we try to pack the required connectors on to the edges of the PCB.

### Notes on Some costs:

1. **MAX232** RS232 level shifter is $0.7335: https://jlcpcb.com/partdetail/TexasInstruments-MAX232D/C350552
2. **DB9 Female** is $3.0222: https://jlcpcb.com/partdetail/TeConnectivity-57478444/C305941

nk25719, 2024-11-27:
https://www.lcsc.com/datasheet/lcsc_datasheet_2005271032_Texas-Instruments-MAX3232EIDR_C193666.pdf

 https://www.ti.com/lit/ml/mpds178g/mpds178g.pdf?ts=1732712591383&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252Fde-de%252FCD4017B

choosing https://jlcpcb.com/partdetail/TexasInstruments-MAX3232EIDR/C193666 : it is availale in high quantity and it is low cost.

[image]

[image]

ForrestErickson, 2024-11-27:
Lee made and committed a new schematic foot print named MAX32_InOrder
[image]

nk25719, 2024-11-28:
Thes are the pinout for the ESP32 UART signals:

[image]

nk25719, 2024-12-01:
Implemented UART1 of the ESP32 through an RS232 line driver and receiver to a DB9 female connector. This is a data communication equipment interface, DCE.

In addition to transmit and receive we added a flow control through RTS and CTS.

[image]

closed 2024-12-01 by nk25719