PubInv/krake #3: Make Wokwi Simulation

ForrestErickson, 2024-05-14:
To the extend possible make a Wokwi simulation of the Krake.

ForrestErickson, 2024-05-14:
Some documentation on Wokwi simulations of the ESP32 platform:
https://docs.wokwi.com/guides/esp32-wifi

There is a ESP32 HTTP IoT Server Example for Wokwi.com
https://wokwi.com/projects/320964045035274834

nk25719, 2024-08-18:
link to Wokwi simulations of Krake: 

https://wokwi.com/projects/406034970467392513

> still processing the simulation..  // Status:  Incomplete

ForrestErickson, 2024-10-02:
Although we can not simulate the WiFi connection with out a paid account a connection by SPI and UART could be some what simulated.  OR CAN IT?

**Item for brief team meeting discussion.**

ForrestErickson, 2024-11-20:
@RobertLRead 
See the information above to be aware of the payment required for WiFi simulation in Wokwi..

At this time I think we can work with out the Wokwi because of the very usefully Hardware2 assembly Nagham designed.

Closing.

closed 2024-11-20 by ForrestErickson