PubInv/krake #5: Add a Rotary Encoder with Switch to Krake

ForrestErickson, 2024-06-01:
# Feature Request:

A rotary encoder will enable development of a more interactive UI.

nk25719, 2024-06-01:
Let us consider the touch screen possibility and compare the advantage and disadvantages of each of these two features.

nk25719, 2024-08-18:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
_Missing hardware that can control the UI of Krake_

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
_Use Rotary knob switches that use a rotating mechanism to select,  send emergency and navigate  and establish connections in the UI of Krake._

ForrestErickson, 2024-10-02:
From the duplicate #20   

History of this issue is that it started in another repository, the **KiCad-esp32-6leds-2switches-1pj-circuit** here:
https://github.com/nk25719/KiCad-esp32-6leds-2switches-1pj-circuit/issues/42

@nk25719 has created firmware at:
krake\Firmware\factoryTest\Rotarty Encoder\RotaryEncoderTest

This firmware interfaces the encoder at the following pins:
> // Define the GPIO pins for CLK, DT, and SW
> const int CLK = 14;   // Connected to the CLK pin of the encoder
> const int DT = 26;    // Connected to the DT pin of the encoder
> const int SW = 25;    // Connected to the SW (Switch) pin of the encoder

Here is a reference to an interrupt driven Rotary Encoder example:
https://www.instructables.com/How-to-Use-Rotary-Encoders-and-Interrupts-With-You/

Note issue #20 now closed.

ForrestErickson, 2024-11-21:
@nk25719 
As of 20241121 The schematic for the Krake has used GPIO25 and GPIO26 on the SPI input.

The Mocking Krake must be modified and rotary encoder firmware to test with the encoder on other pins.

ForrestErickson, 2024-11-23:
@nk25719 
Candidate JLCPCB part: https://jlcpcb.com/partdetail/Alpsalpine-EC11E18244AU/C202365

nk25719, 2024-11-30:
Schematic of the rotary encoder added to Krake: 
[image]

Rotary Encoder properties: 
[image]

nk25719, 2024-11-30:
cc: @ForrestErickson 

.
.

Closing.

nk25719, 2024-12-21:
Reopenning issue to: 

Schematic connections changed, capturing connections Dec 21 2024, 
Implement the Rotary encoder connections into MockingKrake 
Write firmware to test the current Rotary encoder connections before PWA subassembly release.

nk25719, 2024-12-21:
Schematic connections changed, capturing connections Dec 21, 2024.

[image]

nk25719, 2024-12-23:
Connected the Rotery encoder to prototype Krake with it, 
pin connection as above shcematic.

ForrestErickson, 2024-12-24:
As of 20241224 we have added an encoder to the MockingKrake in Maryville and in Lebanon and demonstrated encoder and switch function on the pins as assigned in the Krake schematic.
There is an unresolved firmware problem with double incrementing on the encoder rotation but the hardware works.

Closing.

closed 2024-12-26 by nk25719