PubInv/krake #11: Fault Tolerant Firmware for the loss of WiFi

nk25719, 2024-06-18:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
Loss of WiFi will cause the system to stop opperating.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Indicate the loss of WiFi and automatically reconnect to WiFi when it is available.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
An Icon that is displayed on the LCD that indicates the connectivity to WiFi.

**Additional context**
_Add any other context or screenshots about the feature request here._

nk25719, 2024-08-18:
Krake will constantly check for WiFi connection and incase the connection is lost the Krake will restart and thus we will be submitted back to the wifi manager page where our Krake acts as an AP and allow us to connect to the available WiFi  

```
void loop() {
  
    server.handleClient();

void checkWiFiConnection() {
    if (WiFi.status() != WL_CONNECTED) {
        Serial.println("WiFi lost, reconnecting...");
        WiFiManager wm;
        if (!wm.autoConnect("ESP32_AP")) {
            Serial.println("Failed to connect");
            ESP.restart();
        }
    }
}
```

ForrestErickson, 2024-10-25:
@nk25719 
Two questions

## Have you tested this?
Two ways. Shield or walk away from the WiFi.
Plus another way, to monitor RSSI.

### Sheild
Put in in a stand alone sketch with perhaps the Wink heart beat.  Change the rate of the heart beat if the WiFi connection is true versus false.
Then you must test it by breaking the WiFi connection. How will you do that?  You might try shielding the device under test (the DUT) from RF connection. How will you do that.? Put a metallic shield around the device. How will you do that, Wrap the DUT with some kind of insulator like a paper bag but with a small hole so you can see the LED.. Then around that wrap aluminum foil which is attached to around it but for a small hole so you can see the LED.   
As you close the foil does the device loose WiFi? Open the foil. Does the device recover WiFi?   
Record the results for several trials

### Walk Away.
Put in in a stand alone sketch with perhaps the Wink heart beat.  Change the rate of the heart beat if the WiFi connection is true versus false.  
With the device Battery powered test as follows:
Walk it away from the WiFi router.  Does the device loose WiFi?  
Return to the WiFi router  Does the device recover WiFi?  
Record the results for several trials.  

### Monitor
The WiFi library likely has a function which reports RSSI, Which is an acronym for Received Signal Strength Indicator.
Make a sketch which in addition to a heart beat has a USB-Serial port report and an MQTT publish of the RSSI.  Shield the DUT or walk the DUT away and record the results as above.

## Have you integrated it into some form of the Krake firmware?
If not

nk25719, 2025-07-05:
Integrated OTA and Wifi Manager: 
https://github.com/PubInv/krake/blob/main/Firmware/GPAD_API/WiFiManagerOTA.cpp