PubInv/krake #12: Add A WiFi Manager

nk25719, 2024-06-29:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
When KRAKE moves besisde a new WiFi connection it will not be able to connect to a WIFI.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
WiFi manager will enable KRAKE to connect to different WiFi with different SSID and PASSWORD.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

ForrestErickson, 2024-06-29:
### Important Vocabular, AP (or SoftAP) and STA
As part of understanding the WiFi Management process important vocabular is AP or SoftAP, and STA or Station.

Here is a discussion
https://www.quora.com/What-is-%E2%80%98softAP-and-a-station-in-ESP8266%E2%80%99#:~:text=SoftAP%20means%20Access%20Point%20Mode,network%20specified%20in%20the%20code. 

### Limitation
IF I UNDERSTAND CORRECTLY, I believe there is a fault with the discussion trafic here. It does not make clear that for the ESP the SoftAp and STA modes are mutualy exclusive. You can use one or the other but never both at the same time. 
That is why the AP is really a SoftAP .  An AP connection can provide real internet connectivity. The ESP32 cannot (practically*)

If I am wrong and both modes can work simultaneously, I sure would like to see a reference sketch (code).

## * Practically
It of course might be possible to implement some absurdly limited IP such as being part of this one
https://en.wikipedia.org/wiki/IP_over_Avian_Carriers

ForrestErickson, 2024-06-30:
Regarding, 
Describe the solution you'd like
A clear and concise description of what you want to happen.
WiFi manager will enable KRAKE to connect to different WiFi with different SSID and PASSWORD.

Change To
WiFi manager will enable a user to set a new or additional WiFi with different SSID and PASSWORD.
The user will ?press a button? or ?press a button and hold while applying power? to the KRAKE to enter the manager. 
The user interface on the LCD will advise the user of the status that the KRAKE is in WiFi setup mode for entry of the SSID and PASSWORD.

nk25719, 2024-06-30:
_If I am wrong and both modes can work simultaneously, I sure would like to see a reference sketch (code)._
-------------->>

### **CODE:** 
```

#include <WiFi.h>
#include <WebServer.h>
#include <DNSServer.h>

// Replace with your network credentials
const char* ssidAP = "ESP32-Access-Point";
const char* passwordAP = "123456789";

// Web server running on port 80
WebServer server(80);

// DNS server
const byte DNS_PORT = 53;
DNSServer dnsServer;

// Credentials for connecting to Wi-Fi network
String ssid = "";
String password = "";

void handleRoot() {
  String html = "<html>\
                 <body>\
                 <h1>Enter Wi-Fi Credentials</h1>\
                 <form action=\"/connect\" method=\"POST\">\
                 SSID:<br>\
                 <input type=\"text\" name=\"ssid\"><br>\
                 Password:<br>\
                 <input type=\"password\" name=\"password\"><br><br>\
                 <input type=\"submit\" value=\"Submit\">\
                 </form>\
                 </body>\
                 </html>";
  server.send(200, "text/html", html);
}

void handleConnect() {
  ssid = server.arg("ssid");
  password = server.arg("password");

  server.send(200, "text/html", "<html><body><h1>Connecting...</h1></body></html>");

  WiFi.softAPdisconnect(true);
  WiFi.begin(ssid.c_str(), password.c_str());

  int timeout = 30; // 30 seconds timeout
  while (WiFi.status() != WL_CONNECTED && timeout > 0) {
    delay(1000);
    timeout--;
  }

  if (WiFi.status() == WL_CONNECTED) {
    server.on("/", handleConnectedPage);
    server.send(200, "text/html", "<html><body><h1>Connected! Please navigate to " + WiFi.localIP().toString() + " to access the ESP32.</h1></body></html>");
  } else {
    server.send(200, "text/html", "<html><body><h1>Failed to connect. Please try again.</h1></body></html>");
  }
}

void handleConnectedPage() {
  String html = "<html>\
                 <body>\
                 <h1>ESP32 Connected to Wi-Fi</h1>\
                 <p>IP Address: " + WiFi.localIP().toString() + "</p>\
                 </body>\
                 </html>";
  server.send(200, "text/html", html);
}

void setup() {
  Serial.begin(115200);

  // Set up as Access Point
  WiFi.softAP(ssidAP, passwordAP);
  Serial.println("Access Point Created");

  // Set up DNS server to redirect all URLs to the web server
  dnsServer.start(DNS_PORT, "*", WiFi.softAPIP());

  // Set up web server
  server.on("/", handleRoot);
  server.on("/connect", HTTP_POST, handleConnect);
  server.begin();
  Serial.println("Web Server started");
}

void loop() {
  dnsServer.processNextRequest();
  server.handleClient();
}

```

### **Explanation**
1. Access Point Mode: The ESP32 starts as an access point with the SSID "ESP32-Access-Point" and the password "123456789".
2. Web Server: A web server is set up to serve a web page where you can enter the SSID and password of your Wi-Fi network.
3. Handle Root: The root handler serves the HTML form for entering Wi-Fi credentials.
4. Handle Connect: This handler processes the form submission, attempts to connect to the provided Wi-Fi network, and then serves a success or failure message.
5. Connected Page: If the ESP32 connects to the Wi-Fi network, it serves a new web page with the IP address of the ESP32 on the network.
### 
**Usage**
1. Power on the ESP32: It will create a Wi-Fi network named "ESP32-Access-Point".
2. Connect to this network: Use your phone to connect to the "ESP32-Access-Point" network.
3. Open a browser: Go to http://192.168.4.1 (the default IP for ESP32 in AP mode) to access the web page.
4. Enter Wi-Fi Credentials: Submit the SSID and password of your existing Wi-Fi network.
5. Connect: The ESP32 will attempt to connect to your Wi-Fi network. If successful, it will display the new IP address where you can access it on your network.

> **This setup allows the ESP32 to initially act as an access point, collect Wi-Fi credentials, and then operate as a client on the provided Wi-Fi network, serving a new web page.**

closed 2025-04-24 by ForrestErickson