PubInv/krake #14: Mute Button is not responding

nk25719, 2024-06-29:
**Describe the bug**
_A clear and concise description of what the bug is.
Short Summary of the expected behavior:
Short Summary of the buggy behavior:_
When an emergency alarm should go off when the mute button is pressed,
but the alarm is not going off when we are pressing the mute button 

**To Reproduce**
_Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error_
1- go to the web page and send an emergency alarm by clicking one of the emergency levels
2- try to hold down the mute button on the hardware
3- hear the audio of the alarm, it does not stop

**Expected behavior**
_A clear and concise description of what you expected to happen._
When the mute button is held down the emergency alarm should go off.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware description
Model:
Serial Number:
Other:

**Test Equipment
Model:
Serial Number:

**Additional context**
Add any other context about the problem here.

nk25719, 2024-06-30:
### EXPLANATION: 
Added the mute button firmware to a void function   muteButton(); and calling this function in the void loop() 
Renamed the Button to MUTE_BUTTON_PIN

### CODE:

```
const int MUTE_BUTTON_PIN = 36; // GPIO 36 for the mute button

void setup() {
    Serial.begin(115200);
    pinMode(MUTE_BUTTON_PIN, INPUT_PULLUP); // Set mute button pin as input with pull-up

void loop() {

    muteButton();

void muteButton() {
    int currentState = digitalRead(MUTE_BUTTON_PIN);
    if (lastState == LOW && currentState == HIGH) {
        Serial.println("Mute Button released");
        if (trackPlaying) {
            myDFPlayer.pause();
            trackPlaying = false;
            delay(600000);  // Alarm paused for 10 minutes
            lcd.clear();
            lcd.setCursor(0, 0);
            lcd.print("Alarm Paused");
        } else {
            myDFPlayer.start();
            trackPlaying = true;
            lcd.clear();
            lcd.setCursor(0, 0);
            lcd.print("Alarm Resumed");
        }
        delay(700);
    }
    lastState = currentState;
}
```

nk25719, 2024-06-30:
As of 20240630, 
the mute button is not silencing the alarm when toggled.

ForrestErickson, 2024-11-21:
@nk25719 
Lee thinks this is so stale that we should close it as will not fix. 

If you agree please close.

closed 2024-11-26 by nk25719