PubInv/krake #17: Error at Compile, SoftwareSerial.h: No such file or directory, Krake_arduinoIDE_V2

ForrestErickson, 2024-07-10:
# Summary,

Lee installed the Arduino IED version 2

Lee loaded the sketch Krake_arduinoIDE_V2 and tried to compile. Got error message:

> C:\Users\Public\Downloads\Arduino\PubInv\krake\Krake_Program_V1\Krake_arduinoIDE_V2\Krake_arduinoIDE_V2.ino:5:10: fatal error: SoftwareSerial.h: No such file or directory
>  #include <SoftwareSerial.h>
>           ^~~~~~~~~~~~~~~~~~
> compilation terminated.
> exit status 1
> 
> Compilation error: SoftwareSerial.h: No such file or directory
> 
>

nk25719, 2024-08-18:
> [!CAUTION]
> This happens in the Wokwi simulation only! 

 And this issue is solved in #10 by using the hardware serial of the ESP32

ForrestErickson, 2024-09-29:
Regarding, "And this issue is solved in https://github.com/PubInv/krake/issues/10 by using the hardware serial of the ESP32"
To close this issue all of the firmware in the repository should be modified to remove the requirement for the Softwareserial.h library and migrated to the hardware serial.

The Krake (and the PMD aka KiCad_ESP32_6LED....) uses the UART2 pins

nk25719, 2024-09-30:
[https://github.com/PubInv/krake/blob/main/Firmware/20240421.V2/Krake_arduinoIDE_V2/Krake_arduinoIDE_V2.ino](url)

> path of the modified firmware 

Closing..

ForrestErickson, 2025-04-13:
Reopening.
As of 20250413 the sketch \krake\Firmware\factoryTest\DF_Player\DF_Player.ino still is calling the library not on Lee's machine.

Rewrite DF_Player.ino so that  SoftwareSerial.h is removed.

nk25719, 2025-07-05:
Removed SoftwareSerial.h.

closed 2025-07-05 by nk25719

ForrestErickson, 2025-07-06:
@nk25719 

## Krake
There remains one file in KRAKE still referencing that library (SoftwareSerial.h)
krake\Firmware\20240421.V1\Krake_arduinoIDE_V1\Krake_arduinoIDE_V1.ino

[image]

I think that folder might might be deleted as it is so old as to be of no use.

## Homework
The Homework2 repository has seven references to that library.

[image]