PubInv/krake #20: Prototype a Rotary Encoder

nk25719, 2024-09-07:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
A controller to navigate through the functions of Krake

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
adding an encoder to mocking krake 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

nk25719, 2024-09-07:
https://github.com/nk25719/KiCad-esp32-6leds-2switches-1pj-circuit/issues/42

ForrestErickson, 2024-10-01:
@nk25719 has created firmware at:
krake\Firmware\factoryTest\Rotarty Encoder\RotaryEncoderTest

This firmware interfaces the encoder at the following pins:
> // Define the GPIO pins for CLK, DT, and SW
> const int CLK = 14;   // Connected to the CLK pin of the encoder
> const int DT = 26;    // Connected to the DT pin of the encoder
> const int SW = 25;    // Connected to the SW (Switch) pin of the encoder

ForrestErickson, 2024-10-02:
This appears to be a duplicate to issue #5.

The information in this issue should be copied into #5 and this issue closed as a duplicate.

If we, the team, had been more diligent reviewing issues this duplicate might have been found earlier.  This is symptomatic of not using the project board and issues enough.

ForrestErickson, 2024-10-02:
20241002 Lee copied the content from this issue into the older issue #5

ForrestErickson, 2024-10-02:
Closing as duplicate

closed 2024-10-02 by ForrestErickson

ForrestErickson, 2024-12-24:
20241224 Done.