PubInv/krake #24: Feature Request Learn And Implement "Over The Air Update"

ForrestErickson, 2024-09-17:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

The PMD assemblies will be repurposed in many configurations during development.
We may ship some to persons with out the skill set for Arduino IDE or MSCode loading of firmware.  Learning and implimenting "Over the Air Update" may be helpful.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Teach developers how to use the Over The Air Libraries.  
Load simple blink sketches.  
Load factory test sketches.  
Load PMD sketches.  
Load Krake sketches.  

**Additional context**
_Add any other context or screenshots about the feature request here._  
Random Nerd Tutorials are often a good place to start. 
Here is one on OTA:  https://randomnerdtutorials.com/esp32-over-the-air-ota-programming/

ForrestErickson, 2024-09-17:
Here is an OTA integrated with a WiFi Manager.

https://github.com/erkobg/WiFiManager-OTA

ForrestErickson, 2025-03-18:
## Notes on OTA Development
On 20250317 Nagham and Lee made two sketches with OTA. 
Lee's sketches are not committed. 
Sketch **OTAWebUpdater_Maryville** has a loop with no functions. The log in web page and server (upload) web page are modified to display a header line of  "Best PMD Ever!"
Sketch **OTA_Blink** has a loop function which blinks the LED on pin 2. The log in web page and server (upload) web page are modified to display a header line of  "Best BLINKER Ever!"

When the Arduino IDE "Exports Compiled Binary" files it is into a folder of the sketch with name "..build\esp32.esp32.esp32".
There are several binary files generated. The file with which to update is the file with the sketch name and "..ino.bin". So for the **OTA_Blink** sketch the file is "OTA_Blink.ino.bin".  
[image]

## Notes on OTA behavior.
### Timing of upload to reboot.  
Start with the device is loaded with **OTAWebUpdater_Maryville** the build in LED is not blinking. When the user selects the bin file for "Best BLINKER Ever!" (OTA_Blink.ino.bin)   
[image]

Start timing from the press of the "Update" button. 
[image]
The update becomes 100% in approximately five seconds.
About six seconds later the build in LED begins flashing.

Total time was about 11 to 12 seconds.

ForrestErickson, 2025-03-18:
#  AsyncElegantOTA - A Superior OTA Library and Methods
This RNT article is about a better OTA library. 
ESP32 OTA (Over-the-Air) Updates – AsyncElegantOTA using Arduino IDE  
https://randomnerdtutorials.com/esp32-ota-over-the-air-arduino/

**IMPORTANT:** Note how the HTML , CSS and so on files are easier develop and therefor to manage in the examples.

nk25719, 2025-07-06:
In Krake

[image]

nk25719, 2025-07-06:
Integrated OTA in Krake and in PMD. Closing.

closed 2025-07-06 by nk25719