PubInv/krake #25: Code in V1 has a long "delay" which is dangerous

RobertLRead, 2024-09-19:
Probably, the Arduino freezes completely for 10 minutes in the current code.
Short Summary of the expected behavior:
We would like the "mute" and "unmute" functions to be usable at any time.

Short Summary of the buggy behavior:
Probably when you mute, you can't unmute for 10 minutes.

To Reproduce
Steps to reproduce the behavior:
1. Start playing an alarm.
2. Mute
3. Observe that you cannot unumte for 10 minutes.

Expected behavior
Toggling between muted and unmuted behaviour is possible at any time and that it happens within about 100 ms.

Screenshots
The delay below is likely problematic:

`
    if (trackPlaying) {
        myDFPlayer.pause();
        trackPlaying = false;
        // Allarm paused for 10 minutes
        delay(600000); // 10 minutes delay
        Serial.println("ALLARM Paused.");
        lcd.clear();
        lcd.setCursor(0, 0);
        lcd.print("Alarm Paused");
    } else {
`

Hardware description
This occurs on the MockingKrake/PseudoMedicalDevice (PMD)

Additional context
Nagham now has a better understanding of using timers to avoid long delays.

nk25719, 2024-09-20:
## Exercised these examples in 
https://github.com/nk25719/KiCad-esp32-6leds-2switches-1pj-circuit/tree/main/FIRMWARE

## One of the examples:
FIRMWARE/BlinkWOdelay_jul16c/BlinkWOdelay_jul16c.ino

Solving this in V2
you can find it at: 
https://github.com/PubInv/krake/tree/main/Firmware/20240421.V2

ForrestErickson, 2024-10-02:
Speculation:
The long delays in the code which Lee thinks was inherited from examples found on line my be symptomatic of the DFPlayer requiring long delays between command writes.  The implication is that many DFPLayers require long delays not just the TD5580A based players.

ForrestErickson, 2024-10-04:
Much information about the DFPlayer is to be found in this issue of the PMD aka KiCad-esp32-6leds-2switches-1pj-circuit aka homework2 PCB / PWA

https://github.com/nk25719/KiCad-esp32-6leds-2switches-1pj-circuit/issues/55

ForrestErickson, 2024-10-14:
Here are some easy to read notes about DFPlayers 
https://reprage.com/posts/2018-05-08-dfplayer-mini-cheat-sheet/

## Copying MP3 Files to SD Card
The script after this note looks useful.  

> I had a bunch of MP3 files and micro SD cards to produce, so I made a little script to make it a bit easier (I had all my sequentially numbered files on the desktop in a folder titled ‘soundFiles’):

```
#!/bin/bash
mkdir /Volumes/MUSIC/mp3
for filename in ~/Desktop/soundFiles/*.mp3; do
  cp $filename /Volumes/MUSIC/mp3/
done
dot_clean /Volumes/MUSIC

```

ForrestErickson, 2025-04-25:
Long since removed.
Closing.

closed 2025-04-25 by ForrestErickson