PubInv/krake #26: Software Bug

nk25719, 2024-09-19:
Describe the bug
A clear and concise description of what the bug is.
Short Summary of the expected behavior:
Short Summary of the buggy behavior:
_Symbolic Constants need to be used constanty_ 

To Reproduce
Steps to reproduce the behavior:
Go to Firmware/20240421.V1
Click on -
Scroll down to line 40 
See error 

```
 pinMode(2, OUTPUT);
  pinMode(4, OUTPUT);
  pinMode(5, OUTPUT);
  pinMode(18, OUTPUT);
  pinMode(19, OUTPUT); `

```
Expected behavior
use the lamp initialization on lines 15 to 19

A clear and concise description of what you expected to happen.

Screenshots
If applicable, add screenshots to help explain your problem.
```

const int lamp1Pin = 2;
const int lamp2Pin = 4;
const int lamp3Pin = 5;
const int lamp4Pin = 18;
const int lamp5Pin = 19;

SoftwareSerial mySoftwareSerial(16, 17);  // RX, TX for DFPlayer
DFRobotDFPlayerMini myDFPlayer;
bool trackPlaying = false;
String line;
char command;
unsigned long previousMillis = 0;
const long interval = 100;
LiquidCrystal_I2C lcd(0x3F, 20, 4);  // Address 0x3F, 20 columns, 4 rows
int lastState = LOW; // the previous state (is low) from the input pin
int currentState;     // the current reading from the input pin

void setup() {
  Serial.begin(115200);
  mySoftwareSerial.begin(9600);  // DFPlayer serial communication
  Wire.begin();  // Initialize I2C communication
  lcd.init();  // Initialize LCD
  lcd.backlight();  // Turn on LCD backlight
  pinMode(BUTTON_PIN, INPUT_PULLUP);  // Initialize button pin
  // Initialize lamp pins
  pinMode(2, OUTPUT);
  pinMode(4, OUTPUT);
  pinMode(5, OUTPUT);
  pinMode(18, OUTPUT);
  pinMode(19, OUTPUT);

```

Hardware description
Model:
Serial Number:
Other:

Test Equipment
Model:
Serial Number:

Additional context
Add any other context about the problem here.

ForrestErickson, 2024-10-04:
@nk25719 
Please revisit this bug and edit the subject line from "Software Bug" to something more descriptive of the buggy behavior.

Lee

nk25719, 2025-07-06:
Fixed, closing.

closed 2025-07-06 by nk25719