PubInv/krake #30: Set WiFi Country Code Prior to Shipping

nk25719, 2024-09-30:
Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
_There is difference in Wifi protocols from one country to another, and each country has a WiFi country code, 
this is a preset variable that should be defined by the manufacturer in the device testing phase prior to shipping to user._

Describe the solution you'd like
A clear and concise description of what you want to happen.
_As a user, ship the machine to me with the correct country code._

Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

Additional context
Add any other context or screenshots about the feature request here.

ForrestErickson, 2024-09-30:
There is an interesting discussion about country code here: 
https://github.com/espressif/esp-idf/issues/9521

In particular, if true, this:

> [mhdong](https://github.com/mhdong) commented [on Aug 16, 2022](https://github.com/espressif/esp-idf/issues/9521#issuecomment-1217422452)
> Hi @davidzuhn @AxelLin The default country is "01" (world safe mode) , it will not violate regulatory before calling esp_wifi_set_country_code.

ForrestErickson, 2025-11-16:
## Some AI insight

To set the Wi-Fi country code on an ESP32, you can use the esp_wifi_set_country() function if you are using the ESP-IDF framework, or the AT+CWCOUNTRY AT command if you are using ESP-AT firmware.
Using ESP-IDF:
Include the necessary header.
C

    #include "esp_wifi.h"
Define the wifi_country_t structure with the desired country information. This includes the 2-letter ISO country code, the maximum transmit power, and channel information.
C

    wifi_country_t country_config = {
        .cc = "US", // Example: United States
        .schan = 1,
        .nchan = 11,
        .policy = WIFI_COUNTRY_POLICY_AUTO
    };
Call esp_wifi_set_country() with your defined configuration.
C

    esp_err_t err = esp_wifi_set_country(&country_config);
    if (err != ESP_OK) {
        // Handle error
    }
Using ESP-AT firmware:
Send the AT+CWCOUNTRY command over the serial interface to your ESP32.
Specify the 2-letter ISO country code as a parameter. For example, to set the country to the United States:
Code

    AT+CWCOUNTRY="US"
The ESP32 will respond with OK if the command was successful.
Setting the country code is important for ensuring compliance with local Wi-Fi regulations, particularly regarding allowed channels and transmit power levels.