PubInv/krake #32: Add to PCB, Expansion Headders for Most / All of the ESP32 Kit pins.

ForrestErickson, 2024-10-02:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
For development it may be necessary to connect to unused controller pins.
With out some kind of hole in the PCB to connect a probe or a wire this is often difficult or fragle and therefor error prone.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Add expansion header paralleling most or all of the DSP32 kit pins. The header need not be populated at manufacturing time.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

ForrestErickson, 2024-10-02:
Example of headers paralleling the Due controller on the OEDCS in schematic  
[image]

Example of headers paralleling the Due controller on the OEDCS in PCB   
[image]

Example of JLCPCB header parts  
[image]

nk25719, 2024-11-26:
Duplicate of issue #68 

cc: @ForrestErickson

nk25719, 2024-12-01:
Implemented in schematic: 
[image]

Closing.

closed 2024-12-01 by nk25719