PubInv/krake #37: Understanding MQTT for Telemetery

ForrestErickson, 2024-10-07:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
The MQTT protocol is optimized for telemetry, The distance communication of measurement of data.
Here is a "proof of concept" showing only how an ESP32 (a KRAKE) and a Processing Sketch can exchange data via MQTT. 
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Evaluate where a PMD sends data to a KRAKE through MQTT.

Here is a screen shot of a PMD implemented purely in software in Processing. It publishes message to a topic (unique of the a particular Krake).  
The Krake subscribes to that message.
[image]
 
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

Here is code for the Processing software PMD as a publisher
[MQTT_PMD_DTA_TOPIC.zip](https://github.com/user-attachments/files/17272447/MQTT_PMD_DTA_TOPIC.zip)

Here is ESP32 code for a KRAKE subscribing to the PMD
[MQTT_ESP32DevelopmentBoard.zip](https://github.com/user-attachments/files/17272456/MQTT_ESP32DevelopmentBoard.zip)

ForrestErickson, 2024-10-07:
Processing can be downloaded from here: https://processing.org/download

ForrestErickson, 2024-10-07:
Here is the Processing sketch receiving actual data from the LB1 device in Lebanon.

[image]

ForrestErickson, 2024-10-08:
@nk25719 
## New ESP32 Firmware
I have significantly added functionality to the MQTT example firmware here
[MQTT_ESP32DevelopmentBoard_V0.2.zip](https://github.com/user-attachments/files/17299762/MQTT_ESP32DevelopmentBoard_V0.2.zip)

### Summary
The firmware will now 
Has a heart beat LED.
It send an on line MQTT message every ten seconds and It toggles the "Mute" LED at the same time.
When a publisher message fits a particular patter, the other 5 LEDs can be turned all off, all on or one at a time on.

cc: @RobertLRead

ForrestErickson, 2024-10-08:
@nk25719 
## Summary
New Processing program for publishing to the ESP32 running: **MQTT_ESP32DevelopmentBoard_V0.2.zip**

[image]

1. Run the program.  
2. The window above will open.  
3. Click mouse in window.   
4. Press 0-6 on keyboard and observe what happens on the ESP32 board LEDs.
5. Observe in the sketch dialog the messages from PMDs.
6. [image]

### Firmware:
[MQTT_PMD_DTA_TOPIC 20241008.zip](https://github.com/user-attachments/files/17299959/MQTT_PMD_DTA_TOPIC.20241008.zip)

### Processing
Processing can be downloaded from: https://processing.org/download

ForrestErickson, 2024-10-20:
## An MQTT Web Page
[image]

I have now made a web page which publishes through the shiftr.io Broker an MQTT topic to my device serial number: **20240421_USA4** and sets an LED state.  
It sends the same topic as my Processing sketch. In other words the device **20240421_USA4**  does not know the data came from a web page.

### CODE
Here is the HTML:

```
<!DOCTYPE html>
<html lang="en">
<head>
  <meta charset="utf-8">
  <title>Shifter_client_MQTT</title>.
  <script src="https://unpkg.com/mqtt/dist/mqtt.min.js"></script>
</head>
<body>
<h1>MQTT Shiftr.io Client</h1>
Instructions: Press F12 to see the console.
<hr>
<br>
The MQTT client is: <b>mqtt://public:public@public.cloud.shiftr.io</b>

<script>
let client = mqtt.connect('wss://public:public@public.cloud.shiftr.io');
let topic = "PROCESSING_PMD_USA1_DTA_TOPIC_USA_MARYVILLE"
let metric = 'MessageFromProcessing_PMD: 4'

let topic2 = "20240421_USA4"

client.subscribe(topic);
client.subscribe("20240421_USA4");

client.publish(topic, metric);
console.log("Web page Shifter_client_MQTT publised message: ");
console.log(metric);

client.on('message', function (topic2, message2) { // message is Buffer
  console.log("Got a subscription message topic 20240421_USA4: ");
  console.log(message2.toString());
});

client.end();
</script>

</body
</html>

```

ForrestErickson, 2024-10-20:
A tutorial showing how to receive into a web page MQTT.
https://youtu.be/scgt9qK8xvU?si=lFMvynRLVGhz56i1

ForrestErickson, 2024-10-20:
A tutorial and code for a web socket solution for MQTT on a web page.  

https://youtu.be/scgt9qK8xvU?si=lFMvynRLVGhz56i1  
https://highvoltages.co/iot-internet-of-things/how-to-make-mqtt-web-app-using-html-and-javascript/

ForrestErickson, 2024-10-22:
Probably need a topic which is the krake MODEL_SERIALNUMBER_DATA to which the Krake subscribes.
Probably need a topic which is the krake MODEL_SERIALNUMBER_STATUS to which the Krake publishes.

ForrestErickson, 2024-10-23:
Setup of Wireshark to monitor MQTT traffic on a PC.
[image]

ForrestErickson, 2024-11-11:
@nk25719 
@RobertLRead 
Since you have now both experienced at an introductory level how MQTT can work on the ESP32 platform the goal of this issue has been met.
There are features which require further refinement but the goal of this issue has been met.
Incomplete or incorrect features of our particular MQTT implementation will be in new and independent issues.

Closing.
Lee

ForrestErickson, 2025-10-16:
Other things to learn and implement in both Krake and PMDs

Last Will And Testament
Quality of Service

MQTT Topics, Wildcards. See: https://www.hivemq.com/blog/mqtt-essentials-part-5-mqtt-topics-best-practices/

nk25719, 2025-12-06:
cc: @Yukti007

closed 2026-10-13 by nk25719