PubInv/krake #38: Unexpected New Folder "venv/" on Lee's Clone after PIP Install of Python

ForrestErickson, 2024-10-07:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
On 20241007, when doing a git status on his local clone, Lee discovered an unexpected new untracked folder with files
[image]

This folder probably was created when Lee and Nagham were trying to get Lee's computer to be able to run the Python Flask server as part of Krake development.
  
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 Can this folder be deleted?   
If not why does it not appear in the repository already?
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 
 
**Additional context**
_Add any other context or screenshots about the feature request here._

RobertLRead, 2024-10-07:
This should not be removed.  It is a "virtual environment = venv" folder.

In general, Python allows different versions to be run by allowing you to
use a "virtual environment". This allow you to have both Python 2 and
Python 3 running on the same machine.

I do not believe it should be checked into the repository as it is a
"generated" directory, and would not make sense for every user.

However, I am a rather poor "pythonista". We might have to ask an expert
--- or just google it.

ForrestErickson, 2024-10-07:
@RobertLRead 

Regarding, "I do not believe it should be checked into the repository as it is a
"generated" directory, and would not make sense for every user."

I see two issues.  
- One we need to change the .gitignore file appropriately.
- Two, In a README somewhere, we need information to tell users of this repository of the requirement for installing Python and the other dependencies for the FLASK.

nk25719, 2024-10-09:
the .venv folder, which holds the virtual environment, can be moved to a different folder as long as its new path is correctly referenced when you activate it. However, moving it without adjusting the path might lead to issues when activating the environment.

Regarding informing users of the repository about installing Python and the dependencies (for example, for Flask), it indeed serves a similar purpose to telling Arduino users they need specific libraries to compile code. 

Do we need to provide clear, beginner-friendly instructions or expect users to be familiar with these environments?

In my opinion, We need to know our audiance. If the audience is experienced developers, you can provide minimal setup instructions and expect them to understand dependency management. If they're beginners, more detailed, step-by-step instructions might be necessary.

nk25719, 2025-02-08:
The .venv subject is interesting to look at, 
it is aomething that we need @RobertLRead help in, 
virtual environments are not in use in Krake. 
I am adding this to theContinueeing Education project board to proced with as a learning activiy and closing it as done of Krake's issues. 

Closing .

@ForrestErickson

closed 2025-02-08 by nk25719