PubInv/krake #47: Add SPI Interface

nk25719, 2024-10-19:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 Ability to connect peripherals 
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 Add an SPI (Serial Peripheral Interface) to communicate with peripheral devices
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 
 
**Additional context**
_Add any other context or screenshots about the feature request here._

ForrestErickson, 2024-10-22:
### Reference Data sheet for ESP32 S2:
https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf
See section 3.4.2 Serial Peripheral Interface (SPI)

### To Do, Prototype SPI As Peripheral
Figure out which of the four SPI peripherals are available (not already in use) on the ESP32 Dev Kit we have. 
Prototype an SPI Peripheral (formerly Slave) system and control it with an SPI Controller (Formerly Master) and make

ForrestErickson, 2024-10-22:
## To Do Review ESP32 S2 SPI Libraries
Here is one
https://github.com/hideakitai/ESP32SPISlave

## Random Nerd Tutorials ESP32 SPI Tutorial
https://randomnerdtutorials.com/esp32-spi-communication-arduino/

[image]

ForrestErickson, 2024-10-22:
Added Robert to this issue.

@RobertLRead 
The GPAD firmware will need to be modified so that it can compile on both the UNO controller and the ESP32-S2. 
I am assuming a lot of #ifdefine ESP32-S2 else #ifdef Atmega328 will be required.

ForrestErickson, 2024-10-22:
## SPI Pins on ESP32 DEV
Lee loaded and ran the Random Nerd Tutorials sketch to identify SPI pins on his ESP32-S2 Deve Kit 1
[image]

Bottom line
05:56:43.509 -> MOSI: 23
05:56:43.509 -> MISO: 19
05:56:43.509 -> SCK: 18
05:56:43.509 -> SS: 5

@nk25719 
Please review the MockingKrake for the availability of these pins and any conflict with existing pin usage and report to @RobertLRead  and @ForrestErickson

ForrestErickson, 2024-11-20:
@nk25719 
Wiring two Homework2 assemblies together would be a prudent thing to do in a way that reflects the SPI buss connections we have put into the KRAKE schematic.

Then one device is programed as a Controller and one as a Peripheral to verify the hardware connection works.

In addition there should be a Peripheral Interrupt Request, a GPIO output on the Peripheral which goes into an input on the Controller to signal the Peripheral needs attention.  The  Peripheral Interrupt Request needs to be added to the GPAD schematic.

nk25719, 2024-12-06:
changing the pin assignments of the SPI from 
[image]

GPIO26 - COPI
GPIO25 - SPI_SCK
GPIO 33 - nCS
GPIO32  - CIPO

To pins 
[image]

GPIO23 - COPI
GPIO19 -  CIPO
GPIO18 - SPI_SCK
GPIO5 - nCS

ForrestErickson, 2024-12-06:
changing the pin assignments of the LIGHT signals
LIGHT2, LIGHT3 and LIGHT4

From:
[image]

To:

nk25719, 2024-12-07:
[image]

Changing Pin assignment 
S2 connection from GPIO 33 to GPIO 34 
Mute switch from GPIO 34 to GPIO 35

nk25719, 2024-12-07:
Check issue #50 for latest updates for the SPI Interface.

closed 2024-12-07 by nk25719