PubInv/krake #50: SPI Buss Controler at 5V and the ESP32 at 3.3V May Fail

ForrestErickson, 2024-10-20:
**Describe the Failure / Bug**
_A clear and concise description of what the failure / bug is. Describe the problem NOT the fix._

Short Summary of the expected behavior:  Connect an SPI Controller to and SPI Peripheral should not damage parts.

Short Summary of the buggy behavior:  5V controllers for example UNO may over voltage inputs on 3.3V Peripherals for example ESP32
 
Elementary explanation of voltage incompatibility and some level shifting solutions.

https://www.arrow.com/en/research-and-events/articles/using-a-level-shifter-to-integrate-different-voltage-signals

nk25719, 2024-11-19:
[image]

ForrestErickson, 2024-12-03:
@nk25719 
Philips Semiconductor invented the I2C system.
A more detailed understanding of the MOSFET level shifter and the requirements of I2C is found in the Philips Semiconductor documentation. 
When you have time, see: https://cdn-shop.adafruit.com/datasheets/an97055.pdf

ForrestErickson, 2024-12-03:
## Topics

- What is the depletion region of an PN Junction. See: https://en.wikipedia.org/wiki/Depletion_region
- What is a JFET and how does it work. (It works on the principles of a PN Junction). See: https://www.nutsvolts.com/magazine/article/fet_principles_and_circuits_part_1
- Then what are the additional features of the MOSFET over a JFET? See: https://www.tutorialspoint.com/difference-between-jfet-and-mosfet

The AI summary to the question, "how is a mosfet different from a JFET" is insightful too:

> [image]

nk25719, 2024-12-07:
SPI interface with level shifters and ESD protection. 

[image]

closed 2024-12-07 by nk25719