PubInv/krake #56: Is it possible to power a Krake with a solar cell if it is lit only with normal room illumination?

RobertLRead, 2024-10-23:
**Is your feature request related to a problem? Please describe.**
Getting a chord to the Krake to power it is sometimes inconvenient, and the Krake is usually quiet and dark, spending power only on periodic WiFi use.
 
 
**Describe the solution you'd like**
I know very low-power items, such as watches and calculators, can be powered by a solar cell, even if it never goes outside.
 
 
**Describe alternatives you've considered**
A typical off-the-shelf solar cell/battery power combination may be good enough for this, and produces a convenient USB or 12V power that can go.

ForrestErickson, 2024-10-23:
To answer this we need two pieces. 

1. The measured current draw of the a Krake at worset case
2. And to identify and measure the current produced by a candidate solar cell.

To measure number 1 requires connections with a multimeter in line with the power supply of the Krake as well as a representative Krake.

To measure number 2 requires a dummy load and connections with a multimeter in line with the solar cell.

There are techniques for lower power ESP where you keep it asleep for long periods. This is not compatible with monitoring a medical device.

There are alternatives to the WiFi interface. There is Blue Tooth and BLE (Blue Tooth Low Energy) with lower power requirements on the RF (radio frequency) path.  This is a deep rabbet hole.

nk25719, 2024-10-23:
**Describe alternatives you've considered**

1-  Battery: 3.7V, 2500mAh Li-ion.
2- Solar charging module (e.g., TP4056 , Datasheet: []https://www.digikey.com/en/htmldatasheets/production/2049110/0/0/1/tp4056 ).

ForrestErickson, 2024-10-25:
Information on Solar Cell power production  
[image]

Solar Cell Power Under Room Lighting
[image]

The Homework board running code with WiFi, MQTT and all LEDs on draws about 102 to 150 mA
Assume 200mA 
At an input of 12V that is a power of 12 * 0.2 = 2.4 Watts.

Under direct sunlight, For a solar cell making 200 watts / meter we need.  2.4 / 200 = 0.012 meter^2 
This will be a solar cell 110x110 mm. 

Under room light, for a solar cell making 15 watts / meter we need 2.4 / 15 = 0.16 meter^2 
This will be a solar cell 400x400 mm. 

@RobertLRead 
@nk25719 
Please see my estimates above.

Lee

closed 2025-01-12 by nk25719