PubInv/krake #58: Choose USB to UART Chip,  CH340 or CP210x

ForrestErickson, 2024-10-24:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
I want a reliable USB enumeration of the Krake.
I am frustrated when some UNOs or other devices do not enumerate reliably and some times change COM (USB) port numbers

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
I want reliable repetable enumeration when connecting the USB device (Krake) to a host (A PC) 
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 
 
**Additional context**
_Add any other context or screenshots about the feature request here._

ForrestErickson, 2024-10-24:
The OSHWALAB schematic for their dev kit shows us how to connect to the CH340 USB to UART part.
https://oshwlab.com/Zulfahmi27/esp32-devkit-v1

ForrestErickson, 2024-10-24:
Here is a schematic which uses the CP210 USB to UART chip
https://embedded-systems-design.github.io/overview-of-the-esp32-devkit-doit-v1/SchematicsforESP32.pdf

ForrestErickson, 2024-11-17:
### Here is a candidate Silicon Labs part:
https://jlcpcb.com/partdetail/skyworks_siliconLabs-CP2102GMR/C6568
It is a  WQFN-28 package. Very small with leads that are poorly accessible should rework be necessary.
Qty  Unit Price
1+  $1.9570

### Here is a candidate CH340 part:
https://jlcpcb.com/partdetail/wch_jiangsu_Qin_heng-CH340T/C8689
Package: SSOP-20
Qty  Unit Price
1+  $0.6604

ForrestErickson, 2024-11-25:
There is an article here where the author covers 5V vrsus 3.3V opperation of the CH340 chip for correctl electrical inteface for an ESP32.

https://pcbartists.com/design/embedded/esp32-ch340-programmer-schematic/

Bottom line we should review the data sheet for insructions on how to use the CH340 with the 3.3 V UART of the ESP32.

nk25719, 2024-11-26:
Using CH340T chip: 

[image]

nk25719, 2024-11-30:
[image]

cc: @ForrestErickson 

Closing.

closed 2024-11-30 by nk25719