PubInv/krake #59: Crake Image and Crake Call at Wikipedia

ForrestErickson, 2024-10-25:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 Regarding the name of this project, I am left wondering what does a Crake look and sound like?

 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
There is a sharable image and sound at this wikipedia page:  
https://en.wikipedia.org/wiki/Spotted_crake

**Image**  
https://commons.wikimedia.org/wiki/File:Porzana_porzana_6_(Marek_Szczepanek).jpg 
 
**Sound**    
https://en.wikipedia.org/wiki/File:Spotted_Crake_(Porzana_porzana)_(W1CDR0001382_BD3).ogg

nk25719, 2025-01-12:
Convert the oog file above to mp3 and add to our list files, DFplayer.

ForrestErickson, 2025-04-27:
Lee has down loaded the ogg file.
Lee has converted to an MP3 file type.
Lee has placed the MP3 in the Google Drive folder: Krake/SoundSystem
Link: 
https://drive.google.com/drive/u/1/folders/1uwaVFpFQ0uJbVLxVXngsslgTf2tzOCR6

Closing

CC: @RobertLRead

closed 2025-04-27 by ForrestErickson