PubInv/krake #60: Design Key Cap for Homework2 Switches

ForrestErickson, 2024-10-26:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 I want a key cap on the Homework2 assemblies to work with the EnclosureRevC
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
The key cap should fit over the switch key with a blind pocket (blind hole) some what shorter than the key is tall so that the key can be pressed.
Near the bottom of the key cap the outer size of the key should be larger  than the opening in the top. It should then reduce down to a size smaller than the hole for the keys in the top.  This will keep the key cap from falling out of the enclosure.
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 
 
**Additional context**
_Add any other context or screenshots about the feature request here._

ForrestErickson, 2024-10-26:
Here is an OMRON part which probably matches our part with an "H" of 9mm.
B3F-1070, B3F-1072, B3F-1075, B3F-1072-G

[image]

From Digikey I find and select at 3D model.
[image]

ForrestErickson, 2024-10-26:
Note the Homework2 PWA does not have a 3D model for the switches.
[image]
and
[image]

If we could fix this path perhaps we would have the required file.

## Temporary , test, Solution
However the Omron downloaded STEP file above can be added to a single part in the PCB.
[image]

This is the result for adding to one switch.  
[image]

## Proper Solution
Either fix the path to the built in 3D models AND CHECK THAT "H" IS CORRECT, or
add the Omeron file to the PCB footprint library part and safe with a new part number reflecting the Omron part number "B3F-1070"

nk25719, 2024-10-26:
this is a

nk25719, 2024-10-26:
This is a PMD issue and not Krake's.

nk25719, 2024-10-26:
test email.

ForrestErickson, 2024-11-20:
Adding Oyin to this issue.

Here is an OpenSCAD file with a candidate design:
[Switch_key_caps_WIthGirdel.zip](https://github.com/user-attachments/files/17829099/Switch_key_caps_WIthGirdel.zip)

[image]

ForrestErickson, 2024-11-30:
Moved to the upstream of the Homework2 repository at: 
https://github.com/nk25719/PMD___aka-KiCad-esp32-6leds-2switches-1pj-circuit-/issues/83

Closing.

closed 2024-11-30 by ForrestErickson