PubInv/krake #67: MQTT Provides for LWAT (Last Will and Testament) message, The PMD and the Krake should each take full advantage of LWAT

ForrestErickson, 2024-11-11:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
Network connections may fail and so MQTT connections may fail.
The LWAT is a mechanism where when a client goes off line other clients can be alerted by the broker with a LWAT
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Implement and test LWIT for both PMD and Krake.

There is a description of LWAT here:
https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament/
Both the PMD and the Krake should set a LWAT and should be sent to other device subscribing to a publishing client that has gown down.
 
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
Loss of a keep alive message might also be used. Currently the GPAD_API version 0.12 send every 10 seconds a message that it is on line back to the PMD. 
At Processing PMD version 0.7 the PMD does NOT send a keep alive message.
 
**Additional context**
_Add any other context or screenshots about the feature request here._

ForrestErickson, 2024-11-11:
This video here describes the LWAT setup in a general way:  
https://youtu.be/dNy9GEXngoE?si=JgzZvabAk70sSIg-

On about 20241109 Lee working on a branch not pushed to Krake both a Processing PMD and extension to GPAD_API for LWAT but it it not working.  
It may be that developing **in Processing** a PMD and a "Processing Krake" to work out the LWAT may be the fastest way to work.

nk25719, 2025-05-28:
Part of the MQTT library: Capturing LWT in the library to start integrating it in the PMD 

```
  void setWill(const char topic[]) { this->setWill(topic, ""); }
  void setWill(const char topic[], const char payload[]) { this->setWill(topic, payload, false, 0); }
  void setWill(const char topic[], const char payload[], bool retained, int qos);
  void clearWill();

  void setKeepAlive(int keepAlive);
  void setCleanSession(bool cleanSession);
```

closed 2025-05-29 by nk25719