PubInv/krake #68: Add hearder footprint on every ESP32 pin for troubleshooting access.

ForrestErickson, 2024-11-11:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 
 
**Additional context**
_Add any other context or screenshots about the feature request here._

ForrestErickson, 2024-11-11:
Add single in line package header (SIP header) with 0.1" pitch around the ESP32 package to allow access to every pin for troubleshooting and contingency.

Here is an example from another PI project.  
Schematic   
[image]

PCB  
[image]  

3D Render  
[image]

nk25719, 2024-11-26:
[image]

Adding headers for ESP32 module.

nk25719, 2024-12-01:
[image]

closed 2024-12-01 by nk25719