PubInv/krake #70: ESP32 Requires Reset and Boot Button Circuits from the Development Board

ForrestErickson, 2024-11-12:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
The ESP32 programing from the Arduino IDE requires the Reset and Boot circuit with the push buttons as per the ESP32 Development Kit board schematic.

nk25719, 2024-11-26:
Added Reset and Boot button circuit

[image]

nk25719, 2024-12-02:
Added a boot switch 

[image]

cc: @ForrestErickson 

Closing.

closed 2024-12-02 by nk25719