PubInv/krake #71: Unnecessary and Space consuming .zip files for PCB

ForrestErickson, 2024-11-12:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
I did a pull on 20241111 and got a lot of backup files.
There are unne3cessary .zip files committed into the Krake_V1-backups folder.
The .gitignore must be extended to ignore files in this folder.
 
The files that have already been committed should be removed.

Many of the files:   
[image]

ForrestErickson, 2024-11-12:
Full results of the pull:

> $ git pull
> remote: Enumerating objects: 93, done.
> remote: Counting objects: 100% (87/87), done.
> remote: Compressing objects: 100% (32/32), done.
> remote: Total 71 (delta 51), reused 55 (delta 39), pack-reused 0 (from 0)
> Unpacking objects: 100% (71/71), 1.58 MiB | 264.00 KiB/s, done.
> From https://github.com/PubInv/krake
>    b893dc1..b40bab8  main       -> origin/main
> Updating b893dc1..b40bab8
> Fast-forward
>  Firmware/GPAD_API/GPAD_API.ino                     |    21 +-
>  Firmware/GPAD_API/gpad_serial.cpp                  |     3 -
>  .../Krake_V1-2024-05-11_053055.zip                 |   Bin 0 -> 223653 bytes
>  .../Krake_V1-2024-05-11_055911.zip                 |   Bin 0 -> 350357 bytes
>  .../Krake_V1-2024-05-11_130921.zip                 |   Bin 0 -> 350357 bytes
>  .../Krake_V1-2024-06-30_114750.zip                 |   Bin 0 -> 506325 bytes
>  .../Krake_V1-2024-07-05_145951.zip                 |   Bin 0 -> 506325 bytes
>  .../Krake_V1-2024-07-06_024105.zip                 |   Bin 0 -> 506350 bytes
>  .../Krake_V1-2024-07-06_041508.zip                 |   Bin 0 -> 506350 bytes
>  .../Krake_V1-2024-07-06_073717.zip                 |   Bin 0 -> 506350 bytes
>  .../Krake_V1-2024-07-06_074904.zip                 |   Bin 0 -> 506350 bytes
>  .../Krake_V1-2024-07-06_081027.zip                 |   Bin 0 -> 506350 bytes
>  .../Krake_V1-2024-07-07_003448.zip                 |   Bin 0 -> 506360 bytes
>  .../Krake_V1-2024-08-17_100422.zip                 |   Bin 0 -> 506360 bytes
>  .../Krake_V1-2024-08-31_135245.zip                 |   Bin 0 -> 279358 bytes
>  .../Krake_V1-2024-08-31_140138.zip                 |   Bin 0 -> 228263 bytes
>  .../Krake_V1-2024-08-31_141324.zip                 |   Bin 0 -> 284723 bytes
>  .../Krake_V1-2024-08-31_142103.zip                 |   Bin 0 -> 229523 bytes
>  .../Krake_V1-2024-08-31_143256.zip                 |   Bin 0 -> 284869 bytes
>  .../Krake_V1-2024-09-07_113216.zip                 |   Bin 0 -> 229523 bytes
>  .../Krake_V1-2024-09-07_115219.zip                 |   Bin 0 -> 229523 bytes
>  .../Krake_V1-2024-09-14_052741.zip                 |   Bin 0 -> 229523 bytes
>  .../Krake_V1-2024-09-16_113449.zip                 |   Bin 0 -> 229523 bytes
>  .../Krake_V1-2024-09-16_235741.zip                 |   Bin 0 -> 229523 bytes
>  .../Krake_V1-2024-09-19_135044.zip                 |   Bin 0 -> 229523 bytes
>  .../Krake_V1-2024-10-19_153055.zip                 |   Bin 0 -> 229523 bytes
>  .../Krake_V1-2024-10-26_173147.zip                 |   Bin 0 -> 229523 bytes
>  .../Krake_V1-2024-10-27_043133.zip                 |   Bin 0 -> 229523 bytes
>  .../Krake_V1-2024-10-27_052119.zip                 |   Bin 0 -> 229523 bytes
>  Hardware/Krake_V1.kicad_prl                        |    77 +
>  Hardware/PubInv_Symbols_Library.bak                |    98 +
>  Hardware/fp-info-cache                             | 90518 +++++++++++++++++++
>  PCBa_KRAKE/PCBa_KRAKE.kicad_pcb                    |  3882 +-
>  PCBa_KRAKE/PCBa_KRAKE.kicad_prl                    |     3 +-
>  PCBa_KRAKE/PCBa_KRAKE.kicad_sch                    |  2842 +-
>  PCBa_KRAKE/fp-info-cache                           | 90517 ++++++++++++++++++
>  README.md                                          |    24 +
>  37 files changed, 184918 insertions(+), 3067 deletions(-)
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-05-11_053055.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-05-11_055911.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-05-11_130921.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-06-30_114750.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-07-05_145951.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-07-06_024105.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-07-06_041508.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-07-06_073717.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-07-06_074904.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-07-06_081027.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-07-07_003448.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-08-17_100422.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-08-31_135245.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-08-31_140138.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-08-31_141324.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-08-31_142103.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-08-31_143256.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-09-07_113216.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-09-07_115219.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-09-14_052741.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-09-16_113449.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-09-16_235741.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-09-19_135044.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-10-19_153055.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-10-26_173147.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-10-27_043133.zip
>  create mode 100644 Hardware/Krake_V1-backups/Krake_V1-2024-10-27_052119.zip
>  create mode 100644 Hardware/Krake_V1.kicad_prl
>  create mode 100644 Hardware/PubInv_Symbols_Library.bak
>  create mode 100644 Hardware/fp-info-cache
>

nk25719, 2024-11-19:
Added .gitignore 

Closing.

closed 2024-11-19 by nk25719