PubInv/krake #74: Capture All Krake MAC addresses, Lebanon and USA

ForrestErickson, 2024-11-12:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 I'm always frustrated when I do not know the MAC address of the Krakes.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
We need a table of MAC addresses for Krakes made in Lebanon and the USA.

| **Device SN** | **MAC Address** | **Notes** |
|---------------|-----------------|-----------|
| 20240421_USA1 | 3C61053DF08C    |           |
| 20240421_USA2 | 3C6105324EAC    |           |
| 20240421_USA3 | 3C61053DF63C    |           |
| 20240421_USA4 | 10061C686A14    |           |
| 20240421_USA5 | FCB467F4F74C    | Austin TX |
|               |                 |           |
|               |                 |           |
|               |                 |           |
|               |                 |           |
|               |                 |           |
|               |                 |           |

CC:
@RobertLRead 

 
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 
 
**Additional context**
_Add any other context or screenshots about the feature request here._

ForrestErickson, 2024-12-08:
@nk25719 please find and add the MAC for the MockingKrake device to the above list.

ForrestErickson, 2024-12-12:
Version 0.22 of the PMD_PROCESSING_MQTT now reports the MAC of the PMD too.
Here is Lee's computer in Maryville TN.
[image]

nk25719, 2024-12-12:
Here is Nagham's computer in Aley LB.

[image]