PubInv/krake #75: Code hangs if WiFi is not available...

RobertLRead, 2024-11-13:
**Describe the bug**
_A clear and concise description of what the bug is._
We should work with SPI if we don't have WiFi.
Right now, WiFi being wrong hangs forever.
 
 
**To Reproduce**
Steps to reproduce the behavior:
Set a bad password in the code, and it fails and hangs forever.
 
 
**Expected behavior**
I expect a timeout and a message of disconnection, with attempted reconnections on a period basis.
 
 
**Screenshots**
_If applicable, add screenshots to help explain your problem._
 
 
**Hardware description**
Model:
Serial Number: Austin USA5, FCB467F4F74C
Other:
 
**Test Equipment**
Model:
Serial Number:

**Additional context**
_Add any other context about the problem here._

RobertLRead, 2024-11-14:
This has been fixed in version 0.15. I don't think it hangs any more.  However, I am not closing this until @nk25719 and @ForrestErickson have had a chance to test it.

I tested from the live (github pages) webpage, as well as from the serial port. I did not test it by testing the failure of the WiFi (such as by reseting my router.)

ForrestErickson, 2024-11-14:
## Version 0.15 Fails To Compile

## Arduino Arduino IDE 1.8.9
Lee opened Version 0.15 into the Arduino IDE 1.8.9 and found it will not compile.
[image]

> C:\Users\Public\Downloads\Arduino\PubInv\krake\Firmware\GPAD_API\GPAD_API.ino: In function 'void setup()':
> GPAD_API:352:8: error: 'class WiFiClass' has no member named 'STA'
>    WiFi.STA.begin();
>         ^~~
> Multiple libraries were found for "WiFi.h"
>  Used: C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\WiFi
>  Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
> 

### Arduino Arduino IDE 2.3.3
[image]

> C:\Users\Public\Downloads\Arduino\PubInv\krake\Firmware\GPAD_API\GPAD_API.ino: In function 'void setup()':
> C:\Users\Public\Downloads\Arduino\PubInv\krake\Firmware\GPAD_API\GPAD_API.ino:352:8: error: 'class WiFiClass' has no member named 'STA'
>    WiFi.STA.begin();
>         ^~~
> exit status 1
> Compilation error: 'class WiFiClass' has no member named 'STA'
>

closed 2024-11-20 by ForrestErickson