PubInv/krake #76: GPAD_API, Version 0.14, SerialSplash Does Not Display MAC address, Does not Display Topic

ForrestErickson, 2024-11-13:
**Describe the bug**
_A clear and concise description of what the bug is._
Short Summary of the expected behavior:  Expected to see the MAC at: "08:27:01.711 -> Builtin ESP32 MAC Address: "
Expected to see the alarm topic

Short Summary of the buggy behavior: Did not see MAC and didnot see Alarm topic.
[image]
 
 
**To Reproduce**
Steps to reproduce the behavior:
1. Load the GPAD_API Version 0.14 code into an ESP32.
2. Observe the serial monitor
3. Press reset to see the serial splash screen

CC: @nk25719

ForrestErickson, 2024-11-13:
Analysis, The serialSplash(); must be called after the functions that read the MAC and set the topics.

RobertLRead, 2024-11-13:
It is currently called twice to solve this problem; once before the MAC
address is available, and once after.

This is an imperfect solution and inelegant solution. Can you you verify
that I am correct that it looks like this:
[image: Screenshot 2024-11-13 at 11.07.11 AM.png]

20:44:23.233 -> PubInv GPAD: enter command in form CDa (C is a char, D
is a digit)

20:44:23.265 -> MAC:

20:44:23.265 ->

20:44:23.265 -> ===================================

20:44:23.265 -> KRAKE_

20:44:23.265 -> GPAD_API 0.14

20:44:23.265 -> Builtin ESP32 MAC Address:

20:44:23.265 -> Alarm Topic:

20:44:23.265 -> Broker: public.cloud.shiftr.io

20:44:23.265 -> Compiled at: Nov 12 2024 20:43:24

20:44:23.265 -> GNU Affero General Public License, version 3

20:44:23.265 -> ===================================

20:44:23.265 ->

20:44:23.298 ->

20:44:23.298 -> Connecting to WiFi: readfamilynetwork

20:44:23.978 -> .

20:44:23.978 -> WiFi connected

20:44:23.978 -> My mac is fc:b4:67:f4:f7:4c

20:44:23.978 -> MAC as char array: FCB467F4F74C

20:44:23.978 -> MAC:

20:44:23.978 -> FCB467F4F74C

20:44:23.978 -> XXXXXXX

ForrestErickson, 2024-11-13:
Lee's Arduino 1.8.19 IDE shows the following WiFi libraries
[image]

RobertLRead, 2024-11-14:
This has been made better by v. 0.15.  I am not going to close this; I would like @ForrestErickson to test it and close it if he finds it adequate.

I now get the Mac via a different mechanisms that does not depend on having a WiFi connection (note: Lee's code, using a different library version, never had this problem.)

ForrestErickson, 2025-05-22:
GPAD_API has been displaying the MAC address for some time. The topic can be inferred from this.
Closing.

closed 2025-05-22 by ForrestErickson