PubInv/krake #77: Make a DF Player Footprint

nk25719, 2024-11-13:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 the Df player needs a foorprint and maybe a 3d module as well.
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 create a df player module
create a df player 3d module

nk25719, 2024-11-13:
### created a df player foot print 
the pins start from the top left corner and go in a unti clockwise dorection

approx dimensions : 
x-16.5  and y=19.2
and the sd card slot is represented by the silkscreen of measurement x-14.7 and y-14.5

[image]

nk25719, 2024-11-13:
[image]

ForrestErickson, 2024-11-14:
Review the schematic symbol (which I think is bad), the PCB footprint and the 3D model here:
https://www.snapeda.com/parts/DFR0299/DFRobot/view-part/

ForrestErickson, 2024-11-14:
Regarding, "sd card slot is represented by the silkscreen of measurement x-14.7 and y-14.5"
Make the box hollow. Just make an outline.

Additionally the outline for and SD card when inserted should be added so that we can visualize the card location relative to the enclosure.

An SD card STEP file should be added to the 3D model too.

nk25719, 2024-11-14:
link to dfplayer mini datasheet:
 chrome-extension://efaidnbmnnnibpcajpcglclefindmkaj/https://www.farnell.com/datasheets/3161923.pdf?_gl=1*xtvfhr*_gcl_au*MTQ0OTQzNzY5MS4xNzMxNjEwMjMy

nk25719, 2024-11-16:
[image]

nk25719, 2024-11-19:
Path to DFPlayer Footprint: 
[PCBa_KRAKE/PCBa_KRAKE.pretty/DFPlayer_mini_20241116.kicad_mod](url)

Closing.

closed 2024-11-19 by nk25719