PubInv/krake #78: Learning About .cpp and .h files, Writing Library for Arduino

ForrestErickson, 2024-11-14:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 Nagham asked about the role of .cpp and .h files in the GPAD_API program.

 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
We, Robert and Lee need to help Nagham with refences.

This will require explain the role of  the header files and source files in C and C++

Also from Arduino, try reading this: 
https://docs.arduino.cc/learn/contributions/arduino-creating-library-guide/
 

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 
 
**Additional context**
_Add any other context or screenshots about the feature request here._

nk25719, 2024-11-26:
@ForrestErickson  I can share a scanned copy of the notes I took, 
this is done and we should close as complete.

cc: @ForrestErickson

ForrestErickson, 2024-11-26:
Regarding done. Started, yes. Done no.
Not until you have written some .cpp and .h files and worked with them. 

If I recall correctly, the document at: https://docs.arduino.cc/learn/contributions/arduino-creating-library-guide/
leads you through the process with an example(s).

I too should probably do the same exercise. 

We might also generalize the Wink function into a periodic event library where managing an LED is only a special case.

nk25719, 2025-07-06:
https://github.com/nk25719/PMD___aka-KiCad-esp32-6leds-2switches-1pj-circuit-/tree/main/PMD/PMD_WiFiReady_Elegant_migratingtocpph

closing.

closed 2025-07-06 by nk25719