PubInv/krake #79: Code from v 0.15 requires ESP32 espressif  board to version 3.05

RobertLRead, 2024-11-15:
**Describe the bug**

In @ForrestErickson 's Arduino IDE, v.0.15 of the code failed to compile. By looking at included libraries, Rob concluded the new library included in V0.15 of the code must be in the "package" named "esp32 by Espressif Systems".  We learned that Lee was using the Version 2.X series, but Rob was on the latest 3.05 Version. 
 
After Lee upgraded to the 3.0.5 version, he was able to compile. 

 
 
**Screenshots**
[image]

 
 
**Hardware description**
Model: Krake Homework device
Serial Number:  Austin 
Other:

RobertLRead, 2024-11-15:
@nk25719 @ForrestErickson I am assigning to Nagham only as an "For Your Information (FYI)" courtesy in case she has trouble with this. We need to all be on the same library versions, and unfortunately the Arduino IDE makes that somewhat difficult.

Nagham and Lee should close this issue when they can both successfully compile.

nk25719, 2024-11-19:
As of 20241119 Nagham compiled and uploaded software version 0.19

Closing.

closed 2024-11-19 by nk25719