PubInv/krake #82: MQTT Connection Lost & Not Recover, Processing PMD and GPAD_API V0.17

ForrestErickson, 2024-11-17:
**Describe the bug**
_A clear and concise description of what the bug is._
Short Summary of the expected behavior:  MQTT Connection to recover after waking the PC from sleep.
Short Summary of the buggy behavior:  The Processing PMD lost connection and when restarted still not successfully publishing to nor receiving keepalive subscriptions from GPAD_API V0.17 devices.
 
 
**To Reproduce**
Steps to reproduce the behavior:
Was running the Processing PMD on Lee's PC overnight. 
The PC went to sleep
Restart PC
The Processing PMD console reported connection lost (As it has every other day before)
Today 20241117 restarting the program did not restor connection.
The draw window starts up grey but then goes green when the MQTT connection to broker is made.
In the draw windows the background is green because no packets are received from the broker.
[image]

Using Wireshark Lee found the outgoing MQTT topic with messages to the broker for the five USA devices such as KRAKE_20240421_LEB5_ALM
[image]

However there is no traffic back from the broker with expected MQTT topics such as KRAKE_20240421_LEB5_ACK

ForrestErickson, 2024-11-17:
About 8:20 Lee power cycled the router and  now all four devices are connecting to the WiFI.

HOWEVER what I have learned is that when the WiFi connection can not be made the GPAD_API V0.17 gets hung up trying to make an impossible MQTT connection.

RobertLRead, 2024-11-18:
I believe this is fixed in verison 0.19. I institued a fixed number of retries.

closed 2024-11-18 by RobertLRead