PubInv/krake #84: Long Alarm Message Causes GPAD-API Reset, V0.17

ForrestErickson, 2024-11-18:
**Describe the bug**
_A clear and concise description of what the bug is._
Short Summary of the expected behavior:  Send alarm message of any length and display will at worst truncate message.
Short Summary of the buggy behavior: Device reset and displaed the LCD splash screen.
 
 
**To Reproduce**
Steps to reproduce the behavior:
1. Send alarm message like: let message_alarm4 = 'a4Message From p5 on line. abcdefghijklmnopqrstuvwxyz012345678'  //Works
2. Send alarm message like:   let message_alarm4 = 'a4Message From p5 on line. abcdefghijklmnopqrstuvwxyz0123456789' //Causes reset
 
**Screenshots**
_If applicable, add screenshots to help explain your problem._
Testing while learning p5.js
Screen shot of p5.js program.  
[image]

**Hardware description**
Model:  Homework2 assembly with LCD
Serial Number: USA2 with V0.17 firmware GPAD-API
Other:
 
**Test Equipment**
p5.js on line editor.
File:  https://editor.p5js.org/ForrestErickson/sketches/LBrYk35k7

**Additional context**
_Add any other context about the problem here._
This problem might affects GPAD on the UNO and should be tested.

RobertLRead, 2024-11-18:
I believe this is fixed in version 0.19.

closed 2024-11-18 by RobertLRead

ForrestErickson, 2024-11-19:
CC:
@RobertLRead 
@nk25719 

# Notes on Regression testing of V0.19
About 20241119 0810 EDT, Lee programed four devices with V0.19
From the PMD_PROCESSING Sketch
- 20241119_081319 Msg_recd: 3C61053DF08C_ACK -  online, RSSI:-61.00
- 20241119_081324 Msg_recd: 3C6105324EAC_ACK -  online, RSSI:-67.00
- 20241119_081328 Msg_recd: 10061C686A14_ACK -  online, RSSI:-66.00
- 20241119_081328 Msg_recd: 3C61053DF63C_ACK -  online, RSSI:-70.00

## Testing Journal
Using https://editor.p5js.org/ForrestErickson/sketches/LBrYk35k7 I sent both a short message and the long message and confirmed that USA2 received both messages with out resetting.
Screen shot of the p5 program showing the long message sent 
[image]

Send 80 character long message. Still works.
Screen shot of the 80 character message test
[image]

## Test Summary Results, 
Krake with GPAD-API no longer resets on long alarm message.
Passes.