PubInv/krake #86: KRAKE Should Alert User when WiFi goes Off Line, SPI and USB/Serial Too, Firmware GPAD-API V0.19 Does Not.

ForrestErickson, 2024-11-19:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
With the Krake running firmware GPAD-API V0.19 with out a connection to a USB/serial monitor I have no indication when the WiFi and thus the MQTT connection is lost.
The user has no idea alarms are stale. 

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 Something on the LCD and DFPlayer should alert the user of loss of MQTT connection and that the alarm status is stale.
While the WiFi connection has continuous connection and therefor loss of connection detection, stale data could also happen on a disconnected SPI and USB/Serial interface too but we have no keep alive messages . 

 A requirement for a solution to this should be added to the **User Manual** before the completion and closing of this issue
If implemented for SPI and USB/Serial, this becomes a requirement extension for the API and therefor the controlling PMD too.

RobertLRead, 2024-11-19:
I believe this should be addressed by periodically shifting the LCD to
display something like the MQTT status. This is easily codable.

ForrestErickson, 2024-11-19:
Regarding, "periodically shifting the LCD to display something like the MQTT status"
I think the PMD must send periodic messages.
When the KRAKE fails to receive a message in, perhaps twice, the expected period it needs to alert the user the alarm is stale but then redisplay the alarm if the user recalls it through the menu. At this speculation I have opened wide the big can that is alarm management.

RobertLRead, 2024-11-19:
This could be done by writing a library for use on the PMD =client-side
that maintains the alarm status and resends alarms periodically until they
are dismissed. I agree: this gets us into alarm management, and is
independent of the hardware.

An alternative is to give each a alarm an expiration time. This is probably
terrible from an alarm management point of view.

Taking action based on MQTT status is probably easier.