PubInv/krake #91: Hardware, Delete the I2C connector inheretted from GPAD

nk25719, 2024-11-23:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
J105,  I2C connector creates a layout congestion for the I2C bus. There is no use case for this connector for the KRAKE (and there never was for the GPAD either.) Additionally it occupies layout area which we need to repurpose for the rotary encoder. 

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Remove J105 and R105 and R104 and C113 and C114
 
 
**Additional context**
_Add any other context or screenshots about the feature request here._

[image]

nk25719, 2024-12-02:
[image]

highlighting the I2c connections that should be removed.

ForrestErickson, 2024-12-03:
@RobertLRead 
It is important that you **consent** to this change.
Please review. I understand if you want to speak about it too as there is undocumented history and perhaps not common understanding on all our parts.
Lee

RobertLRead, 2024-12-03:
I do consent to this. @ForrestErickson @nk25719 

I feel like I2C is unlikely to be a useful protocol for the Krake; supporting SPI and a serial interface should be enough.

nk25719, 2024-12-14:
From:

[image]

To :

[image]

closed 2024-12-14 by nk25719