PubInv/krake #94: Software, Add serial commands to KRAKE through UART1

nk25719, 2024-12-01:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
KRAKE hardware is being designed for a serial command input on UART1, following request #2 
Firmware GPAD_API version 0.19 has no support for UART 1 
 
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 Extend the software command input through UART1, just like MQTT

ForrestErickson, 2024-12-01:
Note that the UART1 flow control signals have also been implemented in the hardware.

RobertLRead, 2024-12-05:
@ForrestErickson The Random Nerd Tutorial on this subject suggest that different ESP32 versions (or board kit versions?) have different pin assignments for the UART1. They also say that one of the defaults, 9 and 10, can't be used because it is tied to SPI Flash memory (see section: ESP32 Default UART Pins in the link:
[https://randomnerdtutorials.com/esp32-uart-communication-serial-arduino/](https://randomnerdtutorials.com/esp32-uart-communication-serial-arduino/).)

I believe you have made pin assignments on your PCB -- what pins are you using? I need to know this to put them in the software and to test with them.

nk25719, 2024-12-05:
[image: image.png]

[image: image.png]

GPIO 9 RX
GPIO 10 YX

ForrestErickson, 2024-12-05:
The full information from RNT is as follows:

> About UART1 (GPIO 9 and GPIO10) – these GPIOs are connected to the ESP32 SPI flash memory, so you can’t use them like that. To use UART1 to communicate with other devices, you must define different pins using the HardwareSerial library. 

There is a complexity to the ESP32 where functions like UART TX and UART RX can be mapped to other than the default pins. While I have faced the concept with Pin Mapping on other processor types only in the briefest of ways I have never worked with the ESP32 version.   I guess we need to find tutoriasl on the HardwareSerial library and the pin maping.

Here is a screen shot of the ESP32 part of the Krake Schematic as of 2021041205  
[image]

RobertLRead, 2024-12-05:
Okay --- Even you image does not quite answer the question for me.

But the software is "soft" --- I will code in constants which we can change
later.

But studying the image, I see pins marked:

   - GPIO1 : ControllerTX
   - GPIO3 : Controller RX
   - GPIO17: Controller TX2
   - GPIO16: Controller RX2

This does not match labelling the three build-int ESP32: UARTs 0 for the
serial port, UART1 for the next one and UART2 for the third one.  I suggest
we clarify this and improve the net names (is that the correct terminology.)

ForrestErickson, 2024-12-05:
FYI,
There is an example of using the ESP32 HardwareSerial library on the DFPlayer sketches Nagham and I worked with in this folder:
https://github.com/nk25719/PMD___aka-KiCad-esp32-6leds-2switches-1pj-circuit-/tree/main/FactoryTest/FT_DFplayer

How to assign the UART1 to another set of pins is yet to be discovered.

ForrestErickson, 2024-12-05:
Here is some insight on assigning pins:  
[image]

It remains to figure out in the Arduino framework if we have a "uart_set_pin" function / library.

RobertLRead, 2024-12-06:
This has been fixed in version 0.20 of the firmware. I have tested it with an ESP32 AdaFruit HUZZAH Feather (which uses different pins!) as the transmitter. I have added a test file that works on that.  To use with an ESP32 WROOM, it will likely have to use a different PIN. I attach a jumper wire to D17 on that device (which is the TX1 pin), and to the "RX1" female header on the "HOMEWORK #2" device. It effectively does a "walking ones" test.

ForrestErickson, 2024-12-06:
@RobertLRead 
Regarding, " I attach a jumper wire to D17 on that device"
Please identify the GPIO pin number for the UART1 TX and RX.  In the Krake schematic of 20241205 they are on GPIO10 and GPIO09 respectively but may be in conflict with other functions.

ForrestErickson, 2025-01-01:
As of 20250101, Krake PCB has the following pin asignments:

   - GPIO1 : ControllerTX
   - GPIO3 : Controller RX
   - GPIO17: Controller TX2
   - GPIO16: Controller RX2

ForrestErickson, 2025-01-01:
Firmware updated in version 0.23.
Note that there is a #if !defined (HMWK) which is removing the setup in the code which must go away once we have real KRAKE PCBs.

Closing.

commit 75132b05 (2024-12-06): Fix #94, with a test file
closed 2025-01-01 by ForrestErickson