PubInv/krake #99: Create a diagram showing all of the ways that we can communicate with a Krake

RobertLRead, 2024-12-07:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 I believe the developers' manual should document all of the ways (SPI, UART, MQTT, USB) that can be used to communicate with the Krake.
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 I want a nice diagram.
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 This might be a good time to split the User's Manual into the Operator Manual and Developer's Manual.
 
**Additional context**
_Add any other context or screenshots about the feature request here._

RobertLRead, 2024-12-07:
@nk25719 if you think this is a good idea we could plan who should do it.

ForrestErickson, 2024-12-07:
Suggest Change:
From:
>  ... all of the ways (SPI, UART, MQTT, USB) that can be used to communicate with...

to: 
> ... all of the ways (5V SPI, RS232, MQTT, USB) that can be used to communicate with...