PubInv/krake #100: GPAD_API, Ver 0.20, Incorrect LEDS only when DFPlayer Connected

ForrestErickson, 2024-12-08:
**Describe the bug**
_A clear and concise description of what the bug is._
Short Summary of the expected behavior:  With the "PMD_Processing_MQTT" sketch, when I send the the Lights do not flash properly and in the case of a5 not at all.

## Detailed Behavior

### With the DFPlayer connected on the Homework2 assembly, Send alarms from "PMD_Processing_MQTT".
alarm, a5 alarm all FIVE lights (LEDs should participate in the blinking. However they do not light at all. There is a noticeable flash of redraw to the LCD screen.
When sending a4, the first four Lights light but do not blink an the normal rate. They are on for about four seconds, off for about two seconds. There is a noticeable flash of redraw to the LCD screen about two seconds after the Lights come on.
When sending a3, the first three Lights light but do not blink an the normal rate. They are on for about four seconds, off for about two seconds. There is a noticeable flash of redraw to the LCD screen about two seconds after the Lights come on.
When sending a2, the first two Lights light but do not blink an the normal rate. They are on for about four seconds, off for about two seconds. There is a noticeable flash of redraw to the LCD screen about two seconds after the Lights come on.
When sending a1, the first Light lights but do not blink an the normal rate. It is on for about four seconds, off for about two seconds. There is a noticeable flash of redraw to the LCD screen about two seconds after the Light come on.
When sending a1, the first Light lights but do not blink an the normal rate. It is on for about four seconds, off for about two seconds. There is a noticeable flash of redraw to the LCD screen about two seconds after the Light come on.
When sending a0, the Lights are off. I can not tell if theres is a noticeable flash of redraw to the LCD screen because no back light.

#### Screen shot of the Serial Monitor on the DUT after a single a4 command was sent
[image]

#### Screen shot of the Serial Monitor on the DUT after a single a5 command was sent
[image]

### Observation from Screen Shots
From the screen shots it appears that the KRAKE thinks it is getting repeated alarm commands from the Processing PMD.

### With the DFPlayer NOT connected on the Homework2 assembly,
All alarms work normally.

This screen shot shows how the report from the DUT on the Serial Monitor changed when the DFPlayer was disconnected
[image]

  
**Hardware description**
Model:  Homework2 with LCD and DFPLayer
Serial Number: UAS1
Other: Connect serial monitor on USB

ForrestErickson, 2024-12-08:
## Review of the Homework2 Schematic
See that the TX2 and RX2 are connected to the DFPlayer
[image]

UART1 is set for 
#define TXD1 17
#define RXD1 16

[image]

This means that when the DFPlayer is connected traffic to the DFPlayer may be corrupting the behavior.

### Jumper from GPIO17TXD2 to GPIO16RXD2 did NOT replicate the fault behavior. 
Therefor it is not something like like the signal looping back or echoed from the DFPlayer.

### Oscilloscope on TXD1 and RXD1 signals
The TXD1
[image]
The RXD1 is simply always low.

## Is UART 1 versus UART2 Confusion A Problem?

ForrestErickson, 2024-12-08:
## Checking UART 1 traffice.
Here is the UART 1 setup in GPAD_API
[image]

### Try loop back 
Connect with wire TXD1 17  to RXD1 16.
The device behaves normally.

### DFPLayer Oscilliscope on TXD1 17  to RXD1 16.
Connect DFPlayer

The RXD1 16 line is low never changing

The TTXD1 17 has data in bursts as the LEDs come on.
[image]

ForrestErickson, 2024-12-08:
## More DFPlayer Testing Showing UART Traffice Both TX and RX

I sent a0 to the DUT
I now see data on the RX pin as well as the TX pin.
[image]

ForrestErickson, 2025-01-01:
The solution will be to reassign the UART 1 pins. 
See issue #112 
This has been done in GPAD_API Version 0.23 on 20250101

ForrestErickson, 2025-01-01:
Nagham,
Robert,
Please update the firmware in all KRAKE units.

Closing.

nk25719, 2025-01-01:
Not able to reproduce the error, 
A different behavior on LEB1, all LEDs are blinking when sending emergency level 5 to Homework 2, 
homeswork 2 board is connected to LCD, DFPlayer and Speaker, 
running firmware V.0.21

DF Player connected to LEB1:
[image]

nk25719, 2025-07-05:
Different DFplayer were used to assemble units, prevented duplication of issue. 

However Bug fixed, as per LEE "The solution will be to reassign the UART 1 pins.
See issue https://github.com/PubInv/krake/issues/112"

Fixed, Closing.

nk25719, 2025-07-05:
https://github.com/nk25719/PMD___aka-KiCad-esp32-6leds-2switches-1pj-circuit-

closed 2025-07-05 by nk25719