PubInv/krake #103: Software,  I2c address changes

nk25719, 2024-12-10:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
LCD is not displaying text 
 
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 LCD displays text
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
The I2c address in the code is different from the I2c address when running the I2c address scanning code. 
This is because the I2c exrtender we are using in USA is different from the one we are using in Lebanon 
The problem is exclusive in Krake Prototype, for Krake, the I2c extender will be secified and unique in all KRAKEs
Adding a software that scans the I2c address and then put this address in a variable is a perminant solution for Krake prototypes to display text: LEB1,LEB2,LEB3,LEB4,LEB4,LEB5,USA1,USA2,USA3,USA4,USA5.
 
**Additional context**
_Add any other context or screenshots about the feature request here._

in Issue #101 we documented the difference in the I2c extenders.

RobertLRead, 2024-12-19:
Fixed in version 22.

The following code is in GPAD_HAL.h:
`

// Use these to choose the I2C address of LCD
// GPAD device (earlier version of the Krake)
// #define LCD_ADDRESS 0x38 
// Maryville version
// #define LCD_ADDRESS 0x27 
// General (Lebanon) Version
#define LCD_ADDRESS 0x3F 
`

RobertLRead, 2024-12-19:
@nk25719 @ForrestErickson I believe I have fixed this; please see above documentation. Please close this issue after it has been tested.

ForrestErickson, 2024-12-19:
Lee confirms fix.
After selecting the Maryville I2C define, Version 0.22 is now running on all three Hardware2 assemblies and one ESP Kit in Maryville.

Closing

closed 2024-12-19 by ForrestErickson