PubInv/krake #104: Reivew the Expressif

ForrestErickson, 2024-12-10:
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Lets review the "PCB Layout Design" from the manufacturer.
https://docs.espressif.com/projects/esp-hardware-design-guidelines/en/latest/esp32/pcb-layout-design.html 

Note these guide lines are written about BOTH the ESP32 module (which we will use) and the raw ESP32 integrated circuit.
It is important to keep in mind which case they are writing about.

nk25719, 2025-01-13:
While reviewing this document we found this addtional design guidline

[image]

> https://docs.espressif.com/projects/esp-hardware-design-guidelines/en/latest/esp32/pcb-layout-design.html#positioning-a-module-on-a-base-board

nk25719, 2025-01-15:
[image]

> [Changed the PCB edgecut, removing material under the antenna.](https://github.com/PubInv/krake/commit/be3319726b106f8cf996ba0e81fc5187f912ed67)

closed 2025-01-15 by nk25719