PubInv/krake #109: ESP32 UART1 Has Pin Conflict when set on GPIO9 and GPIO10., Move to Open GPIO Pins

ForrestErickson, 2024-12-15:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

Here is a summary from an AI of the problem
AI Overview
[Learn more](https://support.google.com/websearch?p=ai_overviews&hl=en)
ESP32 UART Communication Pins Explained with Example
To set ESP32's UART1 to pins other than GPIO9 and GPIO10, you need to explicitly configure the UART1 pins in your code using the ESP-IDF framework, as GPIO9 and GPIO10 are typically connected to the flash memory and should not be used for standard UART communication; choose any other available GPIO pins to assign as UART1 TX and RX pins instead. 
Key points to remember:
Default pins:
GPIO9 (TX) and GPIO10 (RX) are the default pins for UART1, but they are not recommended for most applications due to their connection to the flash memory.
Configuring in code:
To use different pins for UART1, you need to specify the desired GPIO pins in your code when initializing the UART peripheral.
 

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Here is a summary from AI of code to change the pin numbers

> #include <HardwareSerial.h>
> // Define new pins for UART1
> const int UART1_TX_PIN = 2; 
> const int UART1_RX_PIN = 3;
> 
> void setup() {
>   // Initialize UART1 with custom pins
>   HardwareSerial UART1(UART1_TX_PIN, UART1_RX_PIN);
>   UART1.begin(9600);  // Set baud rate as needed
> }
> 
> void loop() {
>   // ... use UART1 for communication
> }
> 

 **HOWEVER** for Krake we must not use pin 3 as it is used elsewhere.
We probably want to use GPIO7 and GPIO8 as they unused and are near to the other pins we will be routing.

[image]

RobertLRead, 2024-12-16:
I've addressed this by adding a test file:

[https://github.com/PubInv/krake/blob/main/Firmware/TestUARTLoopBack/TestUARTLoopBack.ino](https://github.com/PubInv/krake/blob/main/Firmware/TestUARTLoopBack/TestUARTLoopBack.ino)

As currently compiled, it uses GPIO12 and GPIO13. I just connect them with a wire.

The screen shot shows that below it works:

[image]

The suggested pins are not available on the Dev Kit (but of course may be when design the board.)  

I can't promise this will work with GPIO7 and GPIO8, but this seems to be strong evidence that it will if those pins are not used for some additional purpose.

@ForrestErickson @nk25719 

Lee, I will allow you to close this, although I consider it done until you ask for something else.

ForrestErickson, 2024-12-16:
## Test the test
Lee loaded the TestUARTLoopBack.ino into an ESP32 DevKit1.
Ran the code and watched the serial output. With out a loop back wire the serial monitor only showed that the data was sent.
With a loop back wire the received message was displayed.

## Test with GPIO7 and GPIO8
Also the DEV Kit1 does not have pins 7 and 8.
Regardless, Lee changed the pins to 7 and 8 and compiled and uploaded the code.
The serial monitor shows the system is resetting 
[image]

## Changed for pins 2 and 15 and retest
This works
[image]

## Conclusion, 
Use GPIO 2 and GPIO15 for UART1, the RS232 port on Krake.

ForrestErickson, 2024-12-16:
## Additional insight on GPIO 7, 8 failure as UART
This URL: https://www.electronicshub.org/esp32-pinout/#:~:text=ESP%2DWROOM%2D32%20Pinout,-Let%20us%20start&text=It%20consists%20of%2038%2Dpin,is%20left%20for%20PCB%20Antenna.
explains how not only GPIO 9 and GPIO 10 are used on the internal flash but so to are GPIO 6, 7, 8 and 11.

[image]

## Bottom line
GPIO 2 and 5 are working for UART1.

ForrestErickson, 2024-12-16:
Lee modified the sketch TestUARTLoopBack.ino to test and demonstrate not only UART1 but also UART2 at the same time.
All three UARTs are demonstrated with success in this sketch.

Once the schematic is changed for UART1 pins compatible like this:
> #define TXD1 2
> #define RXD1 15
this issue can be closed

RobertLRead, 2024-12-16:
Well done, Lee!  Good job! Closing.

nk25719, 2024-12-18:
[image]

Schematic is changed for UART1 pins compatible like this:
 TXD1 2
 RXD1 15

--------------------------------
**Two issues as a result:** 
1. internal battery relocation 
2. Renaming the ESP32 pins : GPIO 9, 10, 2, 15. 

cc: @ForrestErickson

ForrestErickson, 2024-12-19:
@nk25719 
The RTS1 and CTS1 pins should also be moved off of the flash memory pins. 
While I have not demonstrated that use of GPIO11 and GPIO6 would make a memory conflict the documentation implies it will.