PubInv/krake #110: Annotate New Parts in Krake Schematic

ForrestErickson, 2024-12-15:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 
Want to add Logically annotated reference designators to the new parts in the Krake schematic not found in the GPAD.
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 
Use the annotation tool intelligently.

ForrestErickson, 2024-12-15:
## First Annotation Attempt 
**Annotationreport20241215_1316.txt**

```
Annotated 100nF as C303.
Annotated 100nF as C504.
Annotated 47uF 16V as C505.
Annotated 100nF as C701.
Annotated 100nF as C702.
Annotated 100nF as C703.
Annotated 100nF as C704.
Annotated 47uF 16V as C705.
Annotated 100nF as C706.
Annotated BAT54S as D301.
Annotated BAT54S as D302.
Annotated BAT54S as D303.
Annotated BAT54S as D304.
Annotated RED 0603 as D502.
Annotated Conn_01x14 as J104.
Annotated Conn_01x10 as J105.
Annotated Conn_01x14 as J106.
Annotated USB_C_Receptacle_USB2.0 as J201.
Annotated DFPlayermini as J501.
Annotated DE9_Receptacle_MountingHoles as J701.
Annotated Speaker as LS501.
Annotated ENCLOSURE_GPAD_VER1 as MF801.
Annotated PCB_GPAD_VER2 as MF802.
Annotated MMBT2222ALT1G as Q206.
Annotated MMBT2222ALT1G as Q207.
Annotated BSS138 as Q302.
Annotated BSS138 as Q303.
Annotated BSS138 as Q304.
Annotated BSS138 as Q305.
Annotated BSS138 as Q401.
Annotated BSS138 as Q402.
Annotated 1R0 as R101.
Annotated 10K as R103.
Annotated 10K as R104.
Annotated 10K as R105.
Annotated 330R as R108.
Annotated 5K1 as R211.
Annotated 5K1 as R212.
Annotated 1R0 as R213.
Annotated 10K as R214.
Annotated 10K as R215.
Annotated 1R0 as R216.
Annotated 4K7 as R313.
Annotated DNI as R314.
Annotated 4K7 as R315.
Annotated 4K7 as R316.
Annotated DNI as R317.
Annotated 4K7 as R318.
Annotated 4K7 as R319.
Annotated DNI as R320.
Annotated 4K7 as R321.
Annotated 4K7 as R322.
Annotated DNI as R323.
Annotated 4K7 as R324.
Annotated 4K7 as R405.
Annotated 4K7 as R406.
Annotated DNI as R407.
Annotated DNI as R408.
Annotated 4K7 as R409.
Annotated 4K7 as R410.
Annotated 330R as R509.
Annotated 330R as R510.
Annotated 330R as R511.
Annotated 1R0 as R512.
Annotated 330R as R513.
Annotated 330R as R701.
Annotated 330R as R702.
Annotated 330R as R703.
Annotated 330R as R704.
Annotated 330R as R705.
Annotated 330R as R706.
Annotated DNI as R707.
Annotated DNI as R708.
Annotated 330R as R709.
Annotated 330R as R710.
Annotated 1R0 as R711.
Annotated SWITCH_TACTILE_SPST-NO_0.05A_24V as S102.
Annotated RotaryEncoder_Switch as SW101.
Annotated SCL as TP403.
Annotated SDA as TP404.
Annotated MAX3232 as U701.
Error: Duplicate items C110

```

ForrestErickson, 2024-12-15:
## First Annotation Attempt 
**Annotationreport20241215_1333.txt**
This clears the error 

Annotated 100nF as C103.
Annotated 100nF as C303.
Annotated 100nF as C504.
Annotated 47uF 16V as C505.
Annotated 100nF as C701.
Annotated 100nF as C702.
Annotated 100nF as C703.
Annotated 100nF as C704.
Annotated 47uF 16V as C705.
Annotated 100nF as C706.
Annotated BAT54S as D301.
Annotated BAT54S as D302.
Annotated BAT54S as D303.
Annotated BAT54S as D304.
Annotated RED 0603 as D502.
Annotated Conn_01x14 as J104.
Annotated Conn_01x10 as J105.
Annotated Conn_01x14 as J106.
Annotated USB_C_Receptacle_USB2.0 as J201.
Annotated DFPlayermini as J501.
Annotated DE9_Receptacle_MountingHoles as J701.
Annotated Speaker as LS501.
Annotated ENCLOSURE_GPAD_VER1 as MF801.
Annotated PCB_GPAD_VER2 as MF802.
Annotated MMBT2222ALT1G as Q206.
Annotated MMBT2222ALT1G as Q207.
Annotated BSS138 as Q302.
Annotated BSS138 as Q303.
Annotated BSS138 as Q304.
Annotated BSS138 as Q305.
Annotated BSS138 as Q401.
Annotated BSS138 as Q402.
Annotated 1R0 as R101.
Annotated 10K as R103.
Annotated 10K as R104.
Annotated 10K as R105.
Annotated 330R as R108.
Annotated 5K1 as R211.
Annotated 5K1 as R212.
Annotated 1R0 as R213.
Annotated 10K as R214.
Annotated 10K as R215.
Annotated 1R0 as R216.
Annotated 4K7 as R313.
Annotated DNI as R314.
Annotated 4K7 as R315.
Annotated 4K7 as R316.
Annotated DNI as R317.
Annotated 4K7 as R318.
Annotated 4K7 as R319.
Annotated DNI as R320.
Annotated 4K7 as R321.
Annotated 4K7 as R322.
Annotated DNI as R323.
Annotated 4K7 as R324.
Annotated 4K7 as R405.
Annotated 4K7 as R406.
Annotated DNI as R407.
Annotated DNI as R408.
Annotated 4K7 as R409.
Annotated 4K7 as R410.
Annotated 330R as R509.
Annotated 330R as R510.
Annotated 330R as R511.
Annotated 1R0 as R512.
Annotated 330R as R513.
Annotated 330R as R701.
Annotated 330R as R702.
Annotated 330R as R703.
Annotated 330R as R704.
Annotated 330R as R705.
Annotated 330R as R706.
Annotated DNI as R707.
Annotated DNI as R708.
Annotated 330R as R709.
Annotated 330R as R710.
Annotated 1R0 as R711.
Annotated SWITCH_TACTILE_SPST-NO_0.05A_24V as S102.
Annotated RotaryEncoder_Switch as SW101.
Annotated SCL as TP403.
Annotated SDA as TP404.
Annotated MAX3232 as U701.
Annotation complete.

ForrestErickson, 2024-12-15:
In addition to practicing the annotation we need to practice the update PCB to find the best path which preserves as much inheritance with out too much.  The optimal inheritance.

nk25719, 2024-12-16:
Renamed C110 to C? and annotated: 
Sorting by X position.
```
Annotated 100nF as C7.
Annotated 100nF as C8.
Annotated 47uF 16V as C9.
Annotated 100nF as C10.
Annotated 100nF as C11.
Annotated 100nF as C12.
Annotated 100nF as C13.
Annotated 47uF 16V as C14.
Annotated 100nF as C15.
Annotated 100nF as C?1.
Annotated BAT54S as D1.
Annotated BAT54S as D2.
Annotated BAT54S as D3.
Annotated BAT54S as D4.
Annotated RED 0603 as D5.
Annotated Conn_01x14 as J1.
Annotated Conn_01x10 as J2.
Annotated Conn_01x14 as J3.
Annotated USB_C_Receptacle_USB2.0 as J4.
Annotated DFPlayermini as J5.
Annotated DE9_Receptacle_MountingHoles as J6.
Annotated Speaker as LS1.
Annotated ENCLOSURE_GPAD_VER1 as MF1.
Annotated PCB_GPAD_VER2 as MF2.
Annotated MMBT2222ALT1G as Q1.
Annotated MMBT2222ALT1G as Q2.
Annotated BSS138 as Q3.
Annotated BSS138 as Q4.
Annotated BSS138 as Q5.
Annotated BSS138 as Q6.
Annotated BSS138 as Q7.
Annotated BSS138 as Q8.
Annotated 1R0 as R2.
Annotated 10K as R3.
Annotated 10K as R4.
Annotated 10K as R5.
Annotated 330R as R6.
Annotated 5K1 as R7.
Annotated 5K1 as R8.
Annotated 1R0 as R9.
Annotated 10K as R10.
Annotated 10K as R11.
Annotated 1R0 as R12.
Annotated 4K7 as R13.
Annotated DNI as R14.
Annotated 4K7 as R15.
Annotated 4K7 as R16.
Annotated DNI as R17.
Annotated 4K7 as R18.
Annotated 4K7 as R19.
Annotated DNI as R20.
Annotated 4K7 as R21.
Annotated 4K7 as R22.
Annotated DNI as R23.
Annotated 4K7 as R24.
Annotated 4K7 as R25.
Annotated 4K7 as R26.
Annotated DNI as R27.
Annotated DNI as R28.
Annotated 4K7 as R29.
Annotated 4K7 as R30.
Annotated 330R as R31.
Annotated 330R as R32.
Annotated 330R as R33.
Annotated 1R0 as R34.
Annotated 330R as R35.
Annotated 330R as R36.
Annotated 330R as R37.
Annotated 330R as R38.
Annotated 330R as R39.
Annotated 330R as R40.
Annotated 330R as R41.
Annotated DNI as R42.
Annotated DNI as R43.
Annotated 330R as R44.
Annotated 330R as R45.
Annotated 1R0 as R46.
Annotated SWITCH_TACTILE_SPST-NO_0.05A_24V as S1.
Annotated RotaryEncoder_Switch as SW1.
Annotated SCL as TP1.
Annotated SDA as TP2.
Annotated MAX3232 as U2.
Annotation complete.
```

nk25719, 2024-12-16:
Annotated for a second time, 

choosing a different numbering option: 

[image]

```
Annotated 100nF as C103.
Annotated 100nF as C303.
Annotated 100nF as C504.
Annotated 47uF 16V as C505.
Annotated 100nF as C701.
Annotated 100nF as C702.
Annotated 100nF as C703.
Annotated 100nF as C704.
Annotated 47uF 16V as C705.
Annotated 100nF as C706.
Annotated BAT54S as D301.
Annotated BAT54S as D302.
Annotated BAT54S as D303.
Annotated BAT54S as D304.
Annotated RED 0603 as D502.
Annotated Conn_01x14 as J104.
Annotated Conn_01x10 as J105.
Annotated Conn_01x14 as J106.
Annotated USB_C_Receptacle_USB2.0 as J201.
Annotated DFPlayermini as J501.
Annotated DE9_Receptacle_MountingHoles as J701.
Annotated Speaker as LS501.
Annotated ENCLOSURE_GPAD_VER1 as MF801.
Annotated PCB_GPAD_VER2 as MF802.
Annotated MMBT2222ALT1G as Q206.
Annotated MMBT2222ALT1G as Q207.
Annotated BSS138 as Q302.
Annotated BSS138 as Q303.
Annotated BSS138 as Q304.
Annotated BSS138 as Q305.
Annotated BSS138 as Q401.
Annotated BSS138 as Q402.
Annotated 1R0 as R101.
Annotated 10K as R103.
Annotated 10K as R104.
Annotated 10K as R105.
Annotated 330R as R108.
Annotated 5K1 as R211.
Annotated 5K1 as R212.
Annotated 1R0 as R213.
Annotated 10K as R214.
Annotated 10K as R215.
Annotated 1R0 as R216.
Annotated 4K7 as R313.
Annotated DNI as R314.
Annotated 4K7 as R315.
Annotated 4K7 as R316.
Annotated DNI as R317.
Annotated 4K7 as R318.
Annotated 4K7 as R319.
Annotated DNI as R320.
Annotated 4K7 as R321.
Annotated 4K7 as R322.
Annotated DNI as R323.
Annotated 4K7 as R324.
Annotated 4K7 as R405.
Annotated 4K7 as R406.
Annotated DNI as R407.
Annotated DNI as R408.
Annotated 4K7 as R409.
Annotated 4K7 as R410.
Annotated 330R as R509.
Annotated 330R as R510.
Annotated 330R as R511.
Annotated 1R0 as R512.
Annotated 330R as R513.
Annotated 330R as R701.
Annotated 330R as R702.
Annotated 330R as R703.
Annotated 330R as R704.
Annotated 330R as R705.
Annotated 330R as R706.
Annotated DNI as R707.
Annotated DNI as R708.
Annotated 330R as R709.
Annotated 330R as R710.
Annotated 1R0 as R711.
Annotated SWITCH_TACTILE_SPST-NO_0.05A_24V as S102.
Annotated RotaryEncoder_Switch as SW101.
Annotated SCL as TP403.
Annotated SDA as TP404.
Annotated MAX3232 as U701.
Annotation complete.
```

nk25719, 2024-12-19:
Closing as done.

closed 2024-12-19 by nk25719