PubInv/krake #111: Identify Speaker for Krake, 8 Ohm,

ForrestErickson, 2024-12-16:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 We need a speaker manufacturer and part number for Krake.
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Would like one from JLCPCB and PCBWay

ForrestErickson, 2024-12-16:
From Amazon
https://www.amazon.com/Speaker-Speakers-Compatible-Loudspeaker-Player/dp/B09YGDQV3Z

[image]

ForrestErickson, 2024-12-16:
Here is the external speaker connection, J601, from the original GPAD schematic audio page

[image]

nk25719, 2024-12-17:
MFR.Part #
GSPK5010TN-8R2W
JLCPCB Part #
C2847983

in stock: 191
[image]

Find more options here: https://jlcpcb.com/parts/2nd/Audio_Products__Vibration_Motors/Speakers_82655

ForrestErickson, 2024-12-17:
This one is also interesting: C5156349
https://jlcpcb.com/partdetail/Fuet-FS32NS0820_H5_8R01/C5156349

[image]

ForrestErickson, 2024-12-17:
And: C2962677
[image]

ForrestErickson, 2024-12-17:
A large speaker: C19191059
[image]

closed 2025-01-12 by nk25719