PubInv/krake #118: Fiemware, Rotary Encoder is double incrementing

nk25719, 2024-12-24:
**Describe the bug**
_A clear and concise description of what the bug is._
Short Summary of the expected behavior:   Proper increments or decrements for each detent. Rotate the rotary encoder by one detent and get one increment 
Short Summary of the buggy behavior: Rotating the rotary encoder by one detent and getting two increments
 
 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to krake\Firmware\factoryTest\Rotarty Encoder\InterruptRotatorESP32
2. Click on upload
3. Scroll down to Serial monitor
4. See error
 
 
**Screenshots**
_If applicable, add screenshots to help explain your problem._

[image]

 
**Hardware description**
Tested on Models: MockingKRAKE USA1 and  MockingKRAKE LEBNA

ForrestErickson, 2024-12-24:
Note I have tried to change the polarity of the rotation by changing the edges of the ISR asseignment with no success.
All combinations below give the same results. Two steps for each encoder detent.

```
  // register interrupt routine
  attachInterrupt(digitalPinToInterrupt(PIN_IN1), checkPosition, CHANGE);
  attachInterrupt(digitalPinToInterrupt(PIN_IN2), checkPosition, CHANGE);

  // register interrupt routine
  attachInterrupt(digitalPinToInterrupt(PIN_IN1), checkPosition, RISING);
  attachInterrupt(digitalPinToInterrupt(PIN_IN2), checkPosition, RISING);

  // register interrupt routine
  attachInterrupt(digitalPinToInterrupt(PIN_IN1), checkPosition, FALLING);
  attachInterrupt(digitalPinToInterrupt(PIN_IN2), checkPosition, FALLING);

//and

  // register interrupt routine
  attachInterrupt(digitalPinToInterrupt(PIN_IN1), checkPosition, RISING);
  attachInterrupt(digitalPinToInterrupt(PIN_IN2), checkPosition, FALLING);

```

ForrestErickson, 2024-12-24:
## Changing Increment
From " encoder = new RotaryEncoder(PIN_IN1, PIN_IN2, RotaryEncoder::LatchMode::TWO03);"

To:  " encoder = new RotaryEncoder(PIN_IN1, PIN_IN2, RotaryEncoder::LatchMode::FOUR0);"
fixes the problem

See results: 
[image]

nk25719, 2025-02-10:
is this fixed? should we close it? @ForrestErickson

ForrestErickson, 2025-02-11:
Regarding, "is this fixed?"

None of the encoder firmware has been integrated into the GPAD_API so I think this must remain open.

RobertLRead, 2025-05-31:
I have tested this; it does seems to have fixed it. I'm closing this as fixed.

closed 2025-07-05 by nk25719