PubInv/krake #120: Schematic and Firmware, Pin Names for Rotaray Encoder, RT_ENC_CLK, RT_ENC_DT, and nENTER

ForrestErickson, 2024-12-24:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 I'm always frustrated when net names on schematics and firmware do not match.
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 
@nk25719 
Please modify the schematic with the net names on GPIO pins for the rotary encoder as per this example
[image]

@RobertLRead 
Please use these pin definitions above on the Firmware HAL.
See the subject line and the schematic screen shot.

nk25719, 2024-12-26:
[image]

the Rotary encoder datasheet does not use the DT CLK and SW pinnames 

https://www.lcsc.com/datasheet/lcsc_datasheet_1809200034_ALPSALPINE-EC11E18244AU_C202365.pdf

nk25719, 2024-12-26:
Decided to use net names that reflects the names of the pins of the encoder instead of the net names suggested by this issue subject line.

[image]

closed 2024-12-26 by nk25719