PubInv/krake #121: GPAD_API Does Not Report if (!client.connected())

ForrestErickson, 2024-12-25:
**Describe the bug**
_A clear and concise description of what the bug is._

Short Summary of the expected behavior:  
  I want to know if and when an MQTT reconnect() happens.
  I want to know even if I do not have a USB/Serial Monitor connection to the DUT.

Short Summary of the buggy behavior:
 The code silently reconnects.

**Expected behavior**
_A clear and concise description of what you expected to happen._
 
In GPAD_API.ino sketch
I want to change to:
```
#if defined(HMWK)
    if (!client.connected()) {
      Serial.print(millis());
      Serial.println(": MQTT Event, !client.connected()");
      //LCD Message on Client Connect.
      //Line 3
      lcd.setCursor(0, 3);
      lcd.print(millis());
      lcd.print(": MQTT reconnecting");
      reconnect();
    }
```

However the lcd is not in scope and getting it there is beyond Lee at this hour of the day.
 
**Screenshots**
_If applicable, add screenshots to help explain your problem._

[image]

CC: 
@nk25719

ForrestErickson, 2024-12-25:
Note: Lee has code on local branch: ReportReconnectOnLCD

nk25719, 2026-05-16:
completed in Firmware V 0.58