PubInv/krake #124: If We Want A Big Speaker, We Need A Bigger Enclosure

ForrestErickson, 2024-12-27:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
 There is no room for a forward facing speaker with the GPAD PCB outline and the GPAD Enclosure
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 
Consider making a larger PCB and larger enclosure
The red outline here shows a speaker and under it the rotary encoder and some new mounting holes.
[image]

 
**Additional context**
_Add any other context or screenshots about the feature request here._

nk25719, 2025-01-12:
This issue should be reviewed by the entire team.

ForrestErickson, 2025-04-26:
## SPL Sound Pressure Level 
Here is a sound pressure measurement of Krake playing all the audio files at volume = 20 into a 28mm speaker hot melt glued into a Krake Enclosure and into a 6" JBL Speaker in a heavy wooden cabinet.

[image]

## Interpreting SPL  
[image]
 

## Measurement Details
Measurements made with Samsung Galaxy 25S 
Using "Sound Meter KUCO Apps"
KRAKE SN US0001
Audio Files: 

### Setup
Krake and Speaker on floor. Carpet over concrete
Smart Phone on floor 1Meter away

[image]

ForrestErickson, 2025-06-30:
About 20250618 in Austin TX Lee demonstrated to Nagham and Robert the advantage of a louder higher effincy speaker event with out an enclosure using this part

[image]

high efficiency speaker Parts-Express link:
https://www.parts-express.com/Z152948311-3-Sony-Shielded-AV-Speaker-8-Ohm-299-548?quantity=1&custcol1=Z152948311%203%22%20Sony%20Shielded%20AV%20Speaker%208%20Ohm&custcol_ava_item=299-548&custcol_ava_incomeaccount=General&custcol_ava_upccode=848864062037&custcol_ava_pickup=F&custcol_disableshopping=F&undefined=.7

ForrestErickson, 2025-07-09:
On 20250709 
Lee requested additional information from Parts-Express.
https://www.parts-express.com/contact-us

> Regarding a speaker I have purchased, please advise mating connector 
> Part:
> https://www.parts-express.com/Z152948311-3-Sony-Shielded-AV-Speaker-8-Ohm-299-548?quantity=1
> 
> 
> Is this connector designed to be used with a particular crossover network for example?
> 
> Is there a mating inline connector?
> 
> What is the manufacturer and manufacturer part number of the connector / contacts?
> 
> I would be happy to speak with an application engineer. My phone is: 865-518-1040
> 
> Regards,
> (Forrest) Lee Erickson

ForrestErickson, 2025-07-10:
20250710 Found 1.5mm connector system on cables at Amazon

[image]

Here is the JST data page:
https://www.jst.com/products/crimp-style-connectors-wire-to-board-type/zh-connector/

Here is the JLCPCB part number for a two pin PCB vertical header (Male pins with locking shell).
https://jlcpcb.com/partdetail/JST-B2B_ZR_3_4_LF_SN/C163033

nk25719, 2025-08-13:
This enclosure with a new speaker tbd should undergo a dropping test to test test its mechanical design robust.

ForrestErickson, 2025-09-27:
Lee has made a FreeCAD model for a 3" speaker.

[image]

The speaker can be purchased at: https://www.parts-express.com/GRS-3FR-4-Full-Range-3-Speaker-4-Ohm-292-436
The model is in Lee's Github at: https://github.com/ForrestErickson/A3inchSpeaker_GRS_3FR

There is an STL model there which can be imported into OpenSCAD.

ForrestErickson, 2025-11-04:
Regarding, 

## Height
> Height difference identified between 3d model made ( https://www.parts-express.com/GRS-3FR-4-Full-Range-3-Speaker-4-Ohm-292-436) and the height of speaker Sony 1-529-483-11 3" Shielded Full-Range Speaker 8 Ohm, Model: Z152948311.

Please say which is which and the exact difference in mm.

##  Gasket Diameter
> Outer gasket diameter : 74mm
> Inner gasket diameter: 70mm

Is this a request or a notation or  something else?

courtneyludick, 2025-11-26:
Speakers for suitability assessment:
option 1: [YELEMEABHB 3 Inch Full Range Audio Speaker 8Ohm 20W HiFi Stereo Home Loudspeaker For Bookshelf Speakers Desktop DIY 2Pcs Bookshelf speaker](https://www.amazon.co.uk/YELEMEABHB-Speaker-Loudspeaker-Bookshelf-Speakers/dp/B0DR58JZ3N?th=1)
option 2: [3" Hi-Fi Full Range Speaker Driver, 8 Ohm, 50W RMS -  FRS 8 2004](https://cpc.farnell.com/visaton/frs-8-2004/speaker-full-range-3-50w-8ohm/dp/LS01667?mckv=sshopping_dc|pcrid|605262956608|kword||match||plid||slid||product|LS01667|pgrid|138313685455|ptaid|pla-2297175813979|&CMP=KNC-GUK-CPC-SHOPPING-9262013734-138313685455-LS01667&s_kwcid=AL!5616!3!605262956608!!!network}!2297175813979!&gad_source=1&gad_campaignid=9262013734&gbraid=0AAAAAD_m6B3YLM4dzcvkqjS2lNX98mnl8&gclid=CjwKCAiA55rJBhByEiwAFkY1QNPFkaNQ31Rv4_PV7KqmaWHPnF_1OCIOctMOPvTDlwPXZVIIkw4siBoCcngQAvD_BwE)
option 3: [Visaton FR 77 8 Ohm, 3 Inch, Full Range Speaker](https://impactaudio.co.uk/products/visaton-fr-77-8-ohm-3-inch-full-range-speaker)
option 4: [SoundLab 3 inch 10W Full Range Round 8 ohm Speaker](https://cricklewoodelectronics.com/3-inch-10W-Full-Range-Round-8-ohm-Speaker.html?gad_source=1&gad_campaignid=483462886&gbraid=0AAAAAD-6BM4J-Sg8UOID7aQvdwge7AjQL&gclid=CjwKCAiA55rJBhByEiwAFkY1QLY4WEnnOaW_airEO-EFeonIg-8R9i1DPxX4MmCmPNJ1brFZOTiZDhoC6FEQAvD_BwE)

RobertLRead, 2026-01-07:
I purchase this for Courtney and it was still pending as of Jan. 7th:

[https://cpc.farnell.com/visaton/frs-8-2004/speaker-full-range-3-50w-8ohm/dp/LS01667](https://cpc.farnell.com/visaton/frs-8-2004/speaker-full-range-3-50w-8ohm/dp/LS01667)

closed 2026-02-09 by nk25719