PubInv/krake #125: Hardware, location of the rotary encoder on the PWA, GPAD Enclosure

nk25719, 2024-12-27:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
The GPAD enclosure does not have a place for a rotary encoder knob.
 
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 Tentitively placing the rotary encoder in the old place where the speaker was.
 
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
 Find a different location.
 
**Additional context**
_Add any other context or screenshots about the feature request here._
[image]

ForrestErickson, 2025-04-22:
On 20250422 @nk25719  imported the PWA STL file into the enclosure file.

nk25719, 2025-07-06:
https://github.com/PubInv/krake/blob/main/enclosure/U_Box_V105_Krake_RevB.3mf

[image]

Closing.

closed 2025-07-06 by nk25719