PubInv/krake #126: Hardware, PCB, Place Components, Rout Traces, Pass ERC

nk25719, 2024-12-28:
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
 Hardware, PCB, Place Components, Rout Traces, Pass ERC

nk25719, 2024-12-28:
The bottom edge of the board is too cluterred, and to place the new RS232 DB9 we must give up the mounting hole H105

[image]

ForrestErickson, 2025-01-07:
## Initial DRC checks.
Lots to work on.

* From  Sat, Jan 4, 2025 at 2:17 PM 
[image]

* From 1/4/2025 3:01 PM
[image]

nk25719, 2025-01-11:
20250111, cleared and dismissed DRC errors and warning, 
Placed components and routed traces.
Refined and improved Silkscreen readability.
[image]

Closing.

closed 2025-01-11 by nk25719